koalazak/dorita980 #104: dock() after stop() is not executed

hokascha, 2020-01-05:
I got an i7 and I'm trying to send it back to it's dock. So I'm using:
```
myRobotViaLocal.stop()
                .then(() => myRobotViaLocal.dock())
                .then(() => myRobotViaLocal.end())
                .catch(console.log);
```

The robot stops but then doesn't go back to it's homebase. Am I doing something wrong?

thorazine-12, 2020-01-06:
I have the same problem - it is happening since recent iRobot firmware update. I know that because it worked properly few months ago.
If you will send second dock() command then robot will come back to its base.

hokascha, 2020-01-06:
Sending a second `dock()` command right after the first doesn't work for me. What works is restarting the script, e.g. renewing the connection to the bot and then sending a `dock()` command.

koalazak, 2020-01-11:
maybe it is a time issue? did you try to wait a second after the stop() to send the dock()?

hokascha, 2020-01-16:
Your're right! Waiting for 10s between `stop()` and `dock()` works!

closed 2020-03-08 by koalazak

warki, 2020-06-14:
During this period roomba is updating the map. I wait for a new state and then send the "dock" command.

Mirarkitty, 2022-10-10:
You can either wait for "isReady" or until status that it's stopped is reported, I think. Then you can send "dock".