koalazak/dorita980 #105: Does Braava use the same API as Roomba?

thorazine-12, 2020-01-06:
I have I7+ and want to buy Braava m6, but don't know if Braava uses the same API?
Is it possible to use dorita980 to control Braava too (clean, stop, dock etc. - only basic commands)?
Thanks for any help with this!

thorazine-12, 2020-01-10:
Guys, just let you know that Braava can be operated with dorita with no problem! :)
Some commands, statuses etc. are different but it was the same with i7+ compared to previous model.
Below copy of braava m6 config:

```
connected
{
  "cmd": "publish",
  "retain": false,
  "qos": 0,
  "dup": false,
  "length": 3680,
  "topic": "$aws/things/2A3E5707267245C9A7D29312A651B874/shadow/update",
  "payload": {
    "state": {
      "reported": {
        "batPct": 100,
        "batteryType": "F12432784",
        "bbchg": {
          "nChatters": 0,
          "nKnockoffs": 0,
          "nLithF": 2,
          "nChgOk": 1,
          "aborts": [
            0,
            16,
            5
          ],
          "smberr": 57344
        },
        "bbchg3": {
          "estCap": 1925,
          "nAvail": 6,
          "hOnDock": 2,
          "avgMin": 43
        },
        "bbmssn": {
          "aCycleM": 84,
          "nMssnF": 0,
          "nMssnC": 0,
          "nMssnOk": 1,
          "aMssnM": 243,
          "nMssn": 1
        },
        "bbnav": {
          "aMtrack": 0,
          "nGoodLmrks": 116,
          "aGain": 16,
          "aExpo": 46
        },
        "bbpause": {
          "pauses": [
            101,
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            0
          ]
        },
        "bbrstinfo": {
          "nNavRst": 0,
          "nMobRst": 0,
          "causes": "0000",
          "nSafRst": 0,
          "safCauses": [
[... 246 lines cut ...]
        },
        "twoPass": false,
        "tz": {
          "ver": 7,
          "events": [
            {
              "dt": 1564675200,
              "off": 120
            },
            {
              "dt": 1572138001,
              "off": 60
            },
            {
              "dt": 1585443601,
              "off": 120
            }
          ]
        },
        "vacHigh": false
      }
    }
  }
}
```

closed 2020-01-11 by koalazak

koalazak, 2020-01-11:
hey @thorazine-12 can you test this PR with all your robots?
https://github.com/koalazak/dorita980/pull/103

It would be very helpful! thank you!