koalazak/dorita980 #109: `getTime` never returning with my e6198

jfly, 2020-01-31:
First off, thank you for the excellent library!

(I am using `dorita980@3.1.7`)

I'm trying to figure out what the clock on my e6198 roomba is set to, and I'm trying to use the [`getTime`](https://github.com/koalazak/dorita980#getTime) method to do so. However, when I try to use it, it just hangs and seems to never respond. Any advice on how to debug further? Thanks!

jfly, 2020-02-02:
For the record, I *do* see a time in the `lastCommand` piece of the json returned by the `getPreferences` method. For example, using `rest980`:

```
➜  ~ curl -s https://.../api/local/config/preferences | jq .lastCommand
{
  "command": "pause",
  "time": 1580682639,
  "initiator": "localApp"
}
```

David-Hari, 2020-03-08:
Most of the "get" methods just wait for a particular parameter to be received. If you look at the code, you'll see that getTime waits for 'utctime'. If it never returns, it's because your robot never sends that parameter in any update.

koalazak, 2020-03-08:
yeah, seems like your robot is not reportin 'utctime' property on its state.
Use `getRobotState()` to see all your robot state, maybe this robot model is returning the time in other properti.
(time in last command is just the time when the command was fired)

closed 2020-03-08 by koalazak