koalazak/dorita980 #110: Unable to get password

3JL, 2020-02-06:
Hi!
Weird question: I am unable to get a password, and yes I have read the other issues on this topic. But there is something stupid: the Roomba (980, firmware version 2.4.6-3) does not "play a series of tones". I have tried connecting at different times, but no success... Any clue what I may be doing wrong?
J

jeremywillans, 2020-02-06:
This is the same process used to connect the roomba to your mobile app - such as from the [irobot faq](https://homesupport.irobot.com/app/answers/detail/a_id/17734/~/setting-up-your-wi-fi-connected-robot-with-the-irobot%C2%AE-home-app)

Perhaps try [resetting](https://homesupport.irobot.com/app/answers/detail/a_id/9087/~/how-do-i-reboot-my-robot-and-force-close-the-irobot%C2%AE-home-app%3F), and if that dosent work [factory resetting](https://homesupport.irobot.com/app/answers/detail/a_id/9047/~/why-factory-reset-your-robot-and-how-to-do-it%3F) your robot?

3JL, 2020-02-09:
Please forgive my stupidity... I have been pressing the Clean button, rather than the Home button :s

closed 2020-02-09 by 3JL