koalazak/dorita980 #114: What do some of these do?

betolley, 2020-05-08:
set Always Finish?  I have issues where I am trying to get the vacum to finish no mater what but it stops on my floor vents all the time.  Setting this didn't do anything for me.

Set Edge Clean?

betolley, 2020-05-09:
Anyone that look at this the AlwaysFinish seems to continue cleaning if you are stopped with a error and tell it to clean again.  If I dont' set alwaysfinish it just makes a chime.

closed 2020-05-09 by betolley

drinfernoo, 2021-06-07:
Edge Clean makes it do a final pass near all the room edges as the last phase of the cleaning mission, as indicated in the iRobot app:
[image]