koalazak/dorita980 #115: i7 Clean multiple Rooms in desired order

warki, 2020-06-14:
I have an addition to [cleanRoom](https://github.com/koalazak/dorita980#cleanRoom).
At least from Firmware Version 3.8.3 it is possible, to set the desired order in which rooms will be cleaned.
Before, I was able to clean multiple rooms in one task, by adding them to the regions array. But unfortunatelly Roomba always started with the nearest room. Now, with the firmware 3.8.3, there is a new property "ordered", which can be set to "1".

A payload to clean three rooms in order foyer, family_room and kitchen in the desired order would be:

```json
{
  "ordered": "1",
  "regions": [
	{
		"region_id": "11",
		"region_name": "Eingang",
		"region_type": "foyer",
	},
	{
		"region_id": "12",
		 "region_name": "Wohnzimmer",
		 "region_type": "family_room",
	},
	{
		"region_id": "8",
		"region_name": "Küche",
		"region_type": "kitchen",
	}
  ],
  "pmap_id": "S8v123456-1asd-1234567",
  "user_pmapv_id": "190113T132021"
}
```

This would be a good addition for the documentation.

warki, 2020-06-14:
See [Pull Request](https://github.com/warki/dorita980/pull/1)

binomialstew, 2020-07-15:
Were you successful in getting cleanRooms to run with this arg format, @warki? I see in firmware 3.8.3, my regions do not have `region_name` or `region_type`. I don't know how I can get the name or region_type because these properties do not appear in my `lastCommand` state. [I have been unsuccessful in using `cleanRooms`](120), and when I use `"ordered": "1"` it doesn't get a response from the m6 at all.

binomialstew, 2020-07-15:
I now have ordered clean working with the new format, but only when using an integer for the `"ordered"` value rather than a string.

warki, 2020-07-16:
> 
> 
> I now have ordered clean working with the new format, but only when using an integer for the `"ordered"` value rather than a string.

Yes, you're right. I have used an integer value and not a string.

koalazak, 2020-07-16:
nice! feel free to open a PR when you are done with the documentation. And please increase the patch part of the version in the package.json file so I can publish the nwe versino to npm too.
thank you@

closed 2020-08-26 by koalazak