koalazak/dorita980 #117: Roomba s9+ rooms

cecton, 2020-06-21:
On the s9+ as of today the rooms are designated as follow:

```
    {
      "region_id": "7",
      "type": "rid",
    }
```

The type seems to always be `rid`. Only `region_id` varies and is a stringified number.

(I'm just putting this for others, you can close this issue)

binomialstew, 2020-07-15:
I also see this format when I check the`lastCommand` state. Unfortunately, [I have been unable to get `cleanRooms` to run](120) using this or the old format with firmware 3.8.3 on a Braava Jet m6.

binomialstew, 2020-07-15:
I was using a string for `"ordered"` when it should have been an integer `1`. This works now, but I also have omitted the `"type"` property in my request args as per @jeremywillans [suggestion](120):
```
const args = {
  "ordered": 1,
  "regions": [
    {
      "region_id": "11"
    },
    {
      "region_id": "1"
    }
  ],
  "pmap_id": "xxxxxxxxxxxxxxxxxxxxxxxxx",
  "user_pmapv_id": "xxxxxxxxxxxxxxx"
};
```

koalazak, 2020-07-16:
this impacts some how what we have in the current README.md about cleanRoom method? we should change something in the readme? sorry I do not own a s9+ to test it.

binomialstew, 2020-07-17:
I think [warki's PR](https://github.com/warki/dorita980/pull/1) takes care of it

cecton, 2020-07-17:
Please, it was an issue dedicated to Roomba S9+ in particular so people can find the information easily :sweat_smile:

closed 2020-07-17 by cecton