koalazak/dorita980 #120: Cleaning selected rooms

binomialstew, 2020-07-15:
Running firmware 3.8.3 on a Braava Jet m6—it appears the expected room format has changed. When checking the robot state `lastCommand`, I get an object with array items in this format:
```
[
  {
      "region_id": "7",
      "type": "rid",
  },
  ...
]
```
Unfortunately, calling `cleanRoom` from using `dorita980` with this format:
```
{
  "regions": [
    {
      "region_id": "7",
      "type": "rid",
    },
    ...
  ]
  "pmap_id": "[my pmap id]",
  "user_pmapv_id": "[my pmapv id]"
}
``` 
This returns `{"ok":null}` but results in an error with the Jet saying "See the app for help". The app shows "Error 32", meaning there is an issue with the "Imprint Smart Map".

I have tried using the old format but have gotten the same result. This may be because I am unable to determine the correct room names/types correspond with my existing IDs or because the Jet no longer accepts this format for cleanRoom args.

I'd be happy to make a PR to address this, but because of my unfamiliarity with this project, some insight into this issue would definitely help.

jeremywillans, 2020-07-15:
Albiet i use rest980 to interface with dorita980, but I have ordered cleaning working with an i7+ using the following - 

Firmware 3.8.3

```
{
  "ordered": 1,
  "pmap_id": "xxxxxxxxxxxxxx",
  "regions": [
    {"region_id": "11"},
    {"region_id": "13"}
  ],
  "user_pmapv_id": "xxxxxxxxxx"
}
```

binomialstew, 2020-07-15:
Thanks a lot, @jeremywillans! Another example I saw had "ordered" as a string and I was thrown off by that. Using `"ordered": 1` works for me.

closed 2020-07-15 by binomialstew