koalazak/dorita980 #121: Unable to get password

ra-pha, 2020-07-17:
Hi,

I know this is a topic which returns quite often and yet I'm lost.

I have a Roomba 671 (WiFi) and try to run the command to get the user ID and password on my Raspberry Pi. I do get the user ID but instead of the password I get the following message: Error getting password. Follow the instructions and try again.

The Roomba 671 has no Home button, only Spot, Clean and Dock. So basically I try to reboot the Roomba (long press spot & dock for 10 seconds) and run the command when Roomba is back online.

I've tried many many times, always the same result. The Roomba app on my iPhone is closed/off when I run the command. I have updated node and npm on my Raspberry as this had been suggested in other similar issues.

Any help would be appreciated.

warki, 2020-07-17:
Is Roomba on the dock?

jeremywillans, 2020-07-17:
In your case, I believe it’s just hold down the dock button

ra-pha, 2020-07-17:
You are my saviour. 

Push the dock button until the WiFi signal turns to dark green and Roomba makes a sound, then get the user id and password and it works. 

Many thanks Jeremy.

closed 2020-07-17 by ra-pha