koalazak/dorita980 #122: Password not reported

Tokn59, 2020-07-24:
Hello, Thank you for this great contribution to the commnity. I managed to get my BLID using npm run getpassword 192.168.1.205 but did not receive the password :
Robot Data:
{
  ver: '3',
  hostname: 'Roomba-3175000001129740',
  robotname: 'Roomba',
  ip: '192.168.1.205',
  mac: 'D0:C5:D3:BF:2D:4B',
  sw: '3.5.62',
  sku: 'R675040',
  nc: 0,
  proto: 'mqtt',
  cap: { ota: 1, eco: 1, svcConf: 1 },
  blid: '3175000001129740'
}

**Error getting password. Follow the instructions and try again.**

Apparently I needed to press the Home button but my model  Roomba (675) does not have a Home button.

Anything else I could try ?

Many thanks for your help !

Kind regards, Tony

jeremywillans, 2020-07-24:
Try holding the dock button instead?

Tokn59, 2020-07-24:
Thx Jeremy, found the solution : Press and Hold "Dock + Spot" simultaneously until you hear a series of beeps. 

However when I enter Host (IP Address)  + BLID + Password in HASS I get the message "Failed to connect, please try again" ...

closed 2020-07-24 by koalazak