koalazak/dorita980 #127: Unable to request Roomba Password

JustAnotherPU, 2020-09-11:
Hello Dear,

I tried to get my Robot (Roomba M6 Jet) username/blid and password, but each time I stumble on this specific error.
Robot are in the same network as the request, also it's on the charging base and in Wifi Pairing mode.

Can you please point me to the good direction ? 
thanks 

/////
      throw er; // Unhandled 'error' event
      ^

Error: connect EHOSTUNREACH 192.168.1.213:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1141:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'EHOSTUNREACH',
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.213',
  port: 8883
}
////

nilsvdb82, 2020-10-10:
I have the same issue, with a new i7  (firmware 3.2.9)

drinfernoo, 2020-11-25:
I also have this:
```bash
C:\Users\David>get-roomba-password 192.168.1.192
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{
  ver: '3',
  hostname: 'Roomba-3178481001416770',
  robotname: 'Kreacher',
  ip: '192.168.1.192',
  mac: 'D0:C5:D3:CD:4C:75',
  sw: 'v2.4.6-3',
  sku: 'R960020',
  nc: 1,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  },
  blid: '3178481001416770'
}
events.js:291
      throw er; // Unhandled 'error' event
      ^

Error: connect ECONNREFUSED 192.168.1.192:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1145:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ECONNREFUSED',
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.1.192',
  port: 8883
}
```

And am unable to connect to my Roomba 960 to get the password. I had done this once before... but now it's not working. I'm attempting to get rest980 working, but not making much headway 😅

MrAlfabet, 2020-12-09:
Same issue.

`Robot Data:
{ ver: '3',
  hostname: 'iRobot-C0826C5D492C413BABBBC69EF55F297E',
  robotname: 'Roebie',
  robotid: 'C0826C5D492C413BABBBC69EF55F297E',
  ip: '192.168.2.178',
  mac: '50:14:79:19:37:B2',
  sw: 'soho+3.10.8+soho-release-rt320+11',
  sku: 's955840',
  nc: 1,
  proto: 'mqtt',
  cap:
   { carpetBoost: 1,
     binFullDetect: 1,
     dockComm: 1,
     edge: 0,
     maps: 3,
     pmaps: 4,
     tLine: 2,
     area: 1,
     eco: 1,
     multiPass: 2,
     pose: 1,
     team: 1,
     pp: 0,
     '5ghz': 1,
     prov: 3,
     sched: 1,
     svcConf: 1,
     ota: 2,
     log: 2,
     tileScan: 1 },
  blid: 'C0826C5D492C413BABBBC69EF55F297E' }
events.js:174
      throw er; // Unhandled 'error' event
      ^

Error: connect ETIMEDOUT 192.168.2.178:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)
Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:91:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:59:3)
    at process._tickCallback (internal/process/next_tick.js:63:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! dorita980@3.1.8 getpassword: `node ./bin/getpassword.js "192.168.2.178"                                                                   `
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the dorita980@3.1.8 getpassword script.
npm ERR! This is probably not a problem with npm. There is likely additional log                                                                   ging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to in                                                                   stall?

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-09T00_11_43_152Z-debug.log
root@getroombaid2:~/dorita980# cat /root/.npm/_logs/2020-12-09T00_11_43_152Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'run',
1 verbose cli   'getpassword',
1 verbose cli   '192.168.2.178' ]
2 info using npm@6.14.4
3 info using node@v10.19.0
4 verbose run-script [ 'pregetpassword', 'getpassword', 'postgetpassword' ]
5 info lifecycle dorita980@3.1.8~pregetpassword: dorita980@3.1.8
6 info lifecycle dorita980@3.1.8~getpassword: dorita980@3.1.8
7 verbose lifecycle dorita980@3.1.8~getpassword: unsafe-perm in lifecycle true
8 verbose lifecycle dorita980@3.1.8~getpassword: PATH: /usr/share/npm/node_modules/npm-lifecycle/node-gyp-bin:/root/dorita980/node_modules/.bin:/sbin:/bin:/usr/sbin:/usr/bin
9 verbose lifecycle dorita980@3.1.8~getpassword: CWD: /root/dorita980
10 silly lifecycle dorita980@3.1.8~getpassword: Args: [ '-c', 'node ./bin/getpassword.js "192.168.2.178"' ]
11 silly lifecycle dorita980@3.1.8~getpassword: Returned: code: 1  signal: null
12 info lifecycle dorita980@3.1.8~getpassword: Failed to exec getpassword script
13 verbose stack Error: dorita980@3.1.8 getpassword: `node ./bin/getpassword.js "192.168.2.178"`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/share/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid dorita980@3.1.8
15 verbose cwd /root/dorita980
16 verbose Linux 5.4.73-1-pve
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "getpassword" "192.168.2.178"
18 verbose node v10.19.0
19 verbose npm  v6.14.4
20 error code ELIFECYCLE
21 error errno 1
22 error dorita980@3.1.8 getpassword: `node ./bin/getpassword.js "192.168.2.178"`
22 error Exit status 1
23 error Failed at the dorita980@3.1.8 getpassword script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

ChrisTTian667, 2020-12-22:
same issues with iRobot i7 running 3.10.8

xdanik, 2020-12-25:
Just setting up our i7 running on v3.2.9.
At the first try I got:
```
xdanik@pi:~ $ get-roomba-password 192.168.25.108
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
undefined
Robot Data:
{
  ver: '3',
  hostname: 'iRobot-E59AA58EF82044D7A47FE7E4F679AA14',
  robotname: 'Robi',
  robotid: 'E59AA58EF82044D7A47FE7E4F679AA14',
  ip: '192.168.25.108',
  mac: '50:14:79:1F:B8:AA',
  sw: 'lewis+3.2.9+lewis-release-rt419+12',
  sku: 'i715840',
  nc: 0,
  proto: 'mqtt',
  cap: {
    binFullDetect: 1,
    dockComm: 1,
    wDevLoc: 2,
    bleDevLoc: 1,
    maps: 3,
    edge: 0,
    pmaps: 2,
    tLine: 2,
    area: 1,
    eco: 1,
    multiPass: 2,
    pose: 1,
    team: 1,
    pp: 0,
    '5ghz': 1,
    prov: 3,
    sched: 1,
    svcConf: 1,
    ota: 2,
    log: 2
  },
  blid: 'E59AA58EF82044D7A47FE7E4F679AA14'
}
node:events:353
      throw er; // Unhandled 'error' event
      ^

Error: connect ECONNREFUSED 192.168.25.108:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1138:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (node:internal/streams/destroy:188:8)
    at emitErrorCloseNT (node:internal/streams/destroy:153:3)
    at processTicksAndRejections (node:internal/process/task_queues:80:21) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.25.108',
  port: 8883
}
```
But simply executing the same command again worked  - I got the password and Roomba started to welcome me.

ladabartos, 2020-12-25:
Hi, my English is bad. So excuse me I use google translator. I have an i7 (3.10.8) and I solved the same problem. In the end, I solved it like this. When pairing, the i7 turns on its own wifi. Sign up for her. Use the ip r command to find out the ip address of the robot. Enter here in get-roomba-password. You will get a password. After a while, the robot exits the pairing mode and connects to your local network. There you can already use the obtained password.

NickWaterton, 2020-12-27:
The robot needs to be already paired and on your network.
ECONNREFUSED means that something else is already connected to the robot (likely the app).

Force close the app, and retry.

maeneak, 2021-01-03:
I had the same problem and force-stop the app didnt work either. Just had to [reboot the robot ](https://homesupport.irobot.com/app/answers/detail/a_id/9087/~/rebooting-or-resetting-your-robot#:~:text=%20Reboot%20Procedures%20%201%20Roomba%C2%AE%20with%20the,will%20see%20small%20previews%20of%20your...%20More%20)and worked fine

fcmarriott, 2021-01-19:
I have similar issue - but it's just not returning the password. Everything else works fine : 

```
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{
  ver: '3',
  hostname: '<hostname-deleted>',
  robotname: 'Simon',
  ip: '192.168.0.12',
  mac: 'DC:F5:05:35:B0:34',
  sw: 'v2.4.6-3',
  sku: 'R980000',
  nc: 0,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    carpetBoost: 1,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  },
  blid: '3192C22092012660'
}
Error getting password. Follow the instructions and try again.
```

I have probably tried to run this 10 times and just cannot get the Roomba to return the password. 
Any hints?

billyjacoby, 2021-01-31:
@fcmarriott I'm having the same issue. Trying to find a work around now. Were you able to figure it out?

drinfernoo, 2021-01-31:
Force close the (any/all) iRobot app, and make sure your vacuum is connected to the same WiFi and "awake". If it's connected to iRobot, it won't work.

mpptx, 2021-02-09:
Same, Roomba 980. The app is closed, but dorita still returns:
```
$ get-roomba-password 192.168.100.5
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
Robot Data:
undefined
node:events:355
      throw er; // Unhandled 'error' event
      ^

Error: connect ECONNREFUSED 192.168.100.5:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1137:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (node:internal/streams/destroy:188:8)
    at emitErrorCloseNT (node:internal/streams/destroy:153:3)
    at processTicksAndRejections (node:internal/process/task_queues:81:21) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.100.5',
  port: 8883
}
$ 
$ get-roomba-password 192.168.100.5
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{
  ver: '3',
  hostname: 'Roomba-3193C20C30919720',
  robotname: 'Roomba 980',
  ip: '192.168.100.5',
  mac: '80:97:36:C5:D1:84',
  sw: 'v2.4.8-44',
  sku: 'R980040',
  nc: 0,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    carpetBoost: 1,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  },
  blid: '3193D70F30919220'
}
Error getting password. Follow the instructions and try again.
```

ahnyerkeester, 2021-02-11:
Just to tack on a little more information, I have a 680 and am having the same problem getting the password. Same error message so I did a little monkeying with the script to see what we get.
At line number 95 I changed:
```client.on('data', function (data) {
    if (data.length === 2) {
      sliceFrom = 9;
      return;
    }
    if (data.length <= 7) {
      console.log('Error getting password. Follow the instructions and try again.');
```
To
```client.on('data', function (data) {
    if (data.length === 2) {
      sliceFrom = 9;
      return;
    }
    if (data.length <= 0) {
      console.log('Error getting password. Follow the instructions and try again.');
```
And what it returns is:
`Password=>  <= Yes, all this string.`

When I change 
`console.log('Password=> ' + new Buffer(data).slice(sliceFrom).toString() + ' <= Yes, all this string.');`
to
`console.log('Password=> ' + new Buffer(data).toString() + ' <= Yes, all this string.');`

It just puts about three odd characters in there. Looks like V3 of the firmware change how the password is reported.

mpptx, 2021-02-11:
Turned out it was my fault, I was holding CLEAN instead of HOME button - after holding HOME button, the script returned valid password which is perfectly usable in further operations.

ahnyerkeester, 2021-02-11:
Yep, I'm missed it too. So the whole "Follow the instructions and try again" is really good advice!

dylanmcdermott, 2021-02-13:
This is not working for me either. Roomba version per iRobot app (which is not running when trying to run this is 3.5.62-95. 

`Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
undefined
Robot Data:
undefined
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: connect EHOSTDOWN 192.168.86.214:8883 - Local (192.168.86.83:61885)
    at internalConnect (net.js:921:16)
    at defaultTriggerAsyncIdScope (internal/async_hooks.js:430:12)
    at net.js:1009:9
    at processTicksAndRejections (internal/process/task_queues.js:75:11)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  errno: -64,
  code: 'EHOSTDOWN',
  syscall: 'connect',
  address: '192.168.86.214',
  port: 8883`

ahnyerkeester, 2021-02-14:
Hey @dylanmcdermott, I'm pretty sure you got the IP address wrong.  That 86 subnet doesn't sound right. You might check again. I usually log into my router and check the DHCP table.

dylanmcdermott, 2021-02-14:
@ahnyerkeester i definitely did have a typo in the IP address.... what a waste of a few hours! I got this working once I figured this out. Thanks!

closed 2021-02-14 by koalazak

ahnyerkeester, 2021-02-14:
Rock on, partner! I forgot to push the HOME button and wasted a bunch of time digging into code. :)