koalazak/dorita980 #131: getRobotIP/discovery unreliable? Map and current room from current mission?

1075dn, 2021-02-16:
This package is AMAZING! Kudos!

I have both a Roomba i7 and a Braava Jet M6.
Usually when I call getRobotIP I can only get the IP of one robot, and when I call discovery I get the IP of the other.
Sometimes, both getRobotIP and discover return the data for the same robot. Also, sometimes neither getRobotIP nor discovery return any data at all.
It's a little strange. Is it supposed to work like that? Does anyone have any idea as to why this may happen?

Also, is there a way to get which map and room a robot is cleaning? (I know the iRobot HOME app can.)

Thanks so much for your help!

koalazak, 2021-03-02:
It is not in `cleanMissionStatus` state object? (.getState())
Or emmited in mission event? Please, check that and confirm. Thank you!

1075dn, 2021-03-02:
Thanks so much for answering. I can't check yet because I have a new problem - when a robot hasn't been controlled remotely for a few days - does it stop responding to discovery/getRobotIP? I can't connect anymore... It's really weird, because I can control it now using the iRobot home app, but not dorita980. This happened to me again around two weeks ago, and then two days later it started responding again - I'm not sure why. Do you have any idea why this might happen? Thanks!

mad73923, 2021-05-12:
> Thanks so much for answering. I can't check yet because I have a new problem - when a robot hasn't been controlled remotely for a few days - does it stop responding to discovery/getRobotIP? I can't [...]

Did you check if you have only one active connection (server or app)? The robot only supports 1 connection! This solved this issue for me.

1075dn, 2021-05-12:
I don't believe it was connected to anything else (I force closed the app). Maybe it was connected to the cloud? Does that count as a connection?

mad73923, 2021-05-12:
I can't answer this, since I blocked every outgoing connection for my vacuum in my router firewall settings.

1075dn, 2021-05-12:
Hmm... I don't know if I could try that, as until my project is finished we regularly use the app...

koalazak, 2021-07-05:
hello @1075dn,

can you test if the region is here? (reset your robot if it is not accepting connections please)

> It is not in `cleanMissionStatus` state object? (.getState())
> Or emmited in mission event? Please, check that and confirm. Thank you!

closed 2021-07-05 by koalazak