koalazak/dorita980 #134: Password not reading

JBSullivan, 2021-03-15:
No chance to get the Roomba password - any ideas?

`> dorita980@3.1.8 getpassword c:\iobroker\GLT\node_modules\dorita980
> node ./bin/getpassword.js "xxx.xxx.xxx.xxx"

Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{
  ver: '3',
  hostname: 'Roomba-xxxx-ok-xxxxxxx',
  robotname: 'RobbiTobbi',
  ip: 'xxx.xxx.xxx.xxx',
  mac: 'xx:xx:xx:xxx:xx',
  sw: 'v2.4.8-44',
  sku: 'R960040',
  nc: 0,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  },
  blid: 'xxxx-ok-xxxxxxx'
}
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: connect ETIMEDOUT xxx.xxx.xxx.xxx:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1146:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  errno: -4039,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: 'xxx.xxx.xxx.xxx',
  port: 8883
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! dorita980@3.1.8 getpassword: `node ./bin/getpassword.js "xxx.xxx.xxx.xxx"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the dorita980@3.1.8 getpassword script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\iobroker\GLT\env\npm-cache\_logs\2021-03-15T20_11_20_954Z-debug.log

c:\iobroker\GLT\node_modules\dorita980>`

OoDone, 2021-04-30:
I have the same issue

kmclaugh, 2021-05-31:
I have the same problem

lgrueni, 2021-06-03:
I have the same issue, do you find a solution?

koalazak, 2021-07-05:
hey guys, which node.js verions are you using?

lgrueni, 2021-07-05:
Hi, I have V14.15

mundschenk-at, 2021-07-28:
Same problem with a 980 on v2.4.6-3, so I'm pretty sure it's not the new firmware :(

Nyaran, 2021-08-01:
Hi,

I was having the same problem with node v12 (12.21.0), but I tried with v10 (10.24.1) and it worked!

With the version in mind, I checked the releases notes for node v12 and I saw this: https://nodejs.org/tr/blog/uncategorized/10-lts-to-12-lts/#notable-changes-in-node-js-11-8-0

> tls:
>     tls.connect() takes a timeout option analogous to the net.connect() one. nodejs/node#25517

Then, according to the [documentation](https://nodejs.org/docs/latest-v10.x/api/tls.html), I added `timeout: 5000` to https://github.com/koalazak/dorita980/blob/master/bin/getpassword.js#L91 and tried with v12, v14 and v16 and it worked (I needed a retry on v16, maybe another option is needed with the latest version)

Currently I don't have time to test this well and make a Pull Request, but anyone can take the information on this comment to do it =)

koalazak, 2021-09-13:
folks I just tested the lib in latest version with latest `2.4.16-126` firmware on my roomba980 and it works fine with node 10, 12, 14 and 16.

```
➜  dorita980test node ./node_modules/dorita980/bin/getpassword.js 192.168.1.103
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
Robot Data:
{
  ver: '3',
  hostname: 'Roomba-xxxxxxxxx',
  robotname: 'Dorita',
  ip: '192.168.1.103',
  mac: 'xxxxxxxx',
  sw: 'v2.4.16-126',
  sku: 'R98----',
  nc: 0,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    carpetBoost: 1,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  },
  blid: 'xxxxxxxxx'
}
Password=> :1:xxxxx:xxxxxxxxx <= Yes, all this string.
Use this credentials in dorita980 lib :)
```
most common issues getting your password is pressing the wrong button on roomba or having the mobile application opened. Or firewall problems.

I will add the timeout config suggested by @Nyaran anyway

Will close this ticket

closed 2021-09-13 by koalazak