koalazak/dorita980 #137: Mission status sqft no longer reported

drinfernoo, 2021-06-03:
It appears that `sqft` is no longer updated by the robot in firmware 2.4.16-126. I have a Roomba 960, but am trying to find any statistic that could be used to get this from the robot.

koalazak, 2021-09-13:
`sqft` is reported on my roomba980 with latest firmware `2.4.16-126` inside the `cleanMissionStatus` object.

closed 2021-09-13 by koalazak

drinfernoo, 2021-09-13:
Is it updated? The field exists in my response, but never changes from `0`.

koalazak, 2021-09-13:
you are rigth. it exist but always in 0.

koalazak, 2021-09-13:
I see `bbrun.sqft` too but always in 3333 value. Maybe it is updated every X time. I did not leave the robot cleaning too much time to test this...let me know if `bbrun.sqft` works for you

drinfernoo, 2021-09-13:
I believe that the `bbrun` fields are some type of "historical" data, probably lifetime or average values from the robot.

What initially tripped my suspicion (and led to me opening this issue) was that the Home Assistant integration no longer updated it's "area cleaned" (or "cleaning time") attribute(s). I used this package to analyze my response directly, and found that those two fields never update anymore.