koalazak/dorita980 #138: EHOSTUNREACH on both Roomba and Braava

matteobart, 2021-06-20:
Trying to get my roomba and braava password, but I am unable to get either one. I have been trying both IP addresses and they both fail the same way. 

They are both on, held the home button for two seconds, Roomba is flashing wifi and Braava has the circling light, but neither will connect. I'm thinking this may have something to do with the latest iRobot software updates? 

Braava: 3.14.16
Roomba: 2.4.16-126

```
pi@raspberrypi:~ $ get-roomba-password 192.168.1.13
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
events.js:352
      throw er; // Unhandled 'error' event
      ^

Error: connect EHOSTUNREACH 192.168.1.13:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1148:16)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:82:21) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.13',
  port: 8883
}
```

koalazak, 2021-09-13:
dorita980 works fine with latest `2.4.16-126` firmware. Looks like a connectivity issue on your network. A firewall? bad IPs? do you have the mobile application opened at the same time? (close it)

EHOSTUNREACH means network problem. Closing the ticket

closed 2021-09-13 by koalazak

cryptk, 2021-11-12:
@koalazak while this is technically a "network problem" I felt I should give a little more detail.  It seems that on the newer firmwares, when you put the roomba into "password mode", it disconnects from your wifi and runs it's own wifi network.  Typically your phone will connect to this wifi network and do the password exchange.  I was able to use a laptop with wifi, put the roomba into "password mode", connect to the roomba wifi network from the laptop and then run the get-roomba-password command and this successfully returned the password.

It looks like the current method of getting the password may not work anymore because when the roomba is in the right mode to retrieve a password, it is also running it's own wifi network, and you must be connected to that network in order to retrieve the password.

Luckily, this is an open wifi network, so while it's a bit of a pain, there is a workaround.  It is likely possible to cook up a phone app or something that will do this password retrieval.

FWIW, this does seem to be a fairly recent change to the roomba firmware

matteobart, 2021-11-21:
Decided to give this another try over the weekend. My issue was that the IP were not **reserved** for the Roombas. Not sure why that matters, but after going into my router settings and reserving it, everything worked fine.

Hopefully, I save someone else some troubleshooting.