koalazak/dorita980 #141: receiving no username on manual installation

Matt77CHE, 2021-08-13:
Hi
I was trying to set irobot adapter up - Automatic mode did not work, then I tried manual mode, but I'm receiving only:
(used get-roomba-password <my IP>)

...
Then press any key here...
Robot Data:
undefined
Password=> :1:1628866780:TkDimSYeVamemzXA <= Yes, all this string.
Use this credentials in dorita980 lib :)

No username!
How do I get the username?

Matt77CHE, 2021-08-14:
I think there is no longer any support for this adapter ...  ?!?

koalazak, 2021-09-13:
user is the robot blid. It should be there in the "Robot Data" section. Maybe you are using a non suported node.js version.
Try this code:

```
var dorita980 = require('dorita980');

dorita980.discovery((ierr, data) => {
  console.log(data);
});
```
Should print the robot hostname which contains the robot BLID `Roomba-xxxxxx` (xxxx part is the BLID)

It is really hard to give you support with that lack of information you provide about your setup...

regards

closed 2021-09-13 by koalazak