koalazak/dorita980 #142: J7?

dcmeglio, 2021-09-10:
Anyone tested this with the new j7 yet?

jdiegmueller, 2021-09-14:
> Anyone tested this with the new j7 yet?

I just picked up a Roomba j7+.  However, I can't seem to get it to fetch the password from the device:
```
[jdiegmueller@nv-dc2-rproxy1 dorita980]# get-roomba-password 10.245.2.102
Make sure your robot is on the Home Base and powered on (green lights on). Then press and hold the HOME button (or DOCK+SPOT on some models) on your robot until it plays a series of tones (about 2 seconds). Release the button and your robot will flash WIFI light.
Then press any key here...
events.js:291
      throw er; // Unhandled 'error' event
      ^

Error: Client network socket disconnected before secure TLS connection was established
    at connResetException (internal/errors.js:609:14)
    at TLSSocket.onConnectEnd (_tls_wrap.js:1557:19)
    at Object.onceWrapper (events.js:420:28)
    at TLSSocket.emit (events.js:326:22)
    at endReadableNT (_stream_readable.js:1241:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:21)
Emitted 'error' event on TLSSocket instance at:
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  code: 'ECONNRESET',
  path: undefined,
  host: '10.245.2.102',
  port: 8883,
  localAddress: undefined
}

```

I've also tried the password-fetching tools from the similar(?) project (https://github.com/NickWaterton/Roomba980-Python) with similar results, so I suspect there is something different about the j7+.

koalazak, 2021-09-14:
hello folks,

what do you see if run this snippet? (make sure you are in the same network as the robot!)

```javascript
const dorita980 = require('dorita980');

dorita980.discovery((ierr, data) => {
  console.log(data);
});
```

jdiegmueller, 2021-09-15:
[root@ROCKY dorita980]# ./jason.js 
Looking for robots...
Robot found! with blid/username: BCBCDD45160547239EBF3D8B086CB7F2
{ ver: '3',
  hostname: 'iRobot-BCBCDD45160547239EBF3D8B086CB7F2',
  robotname: 'Braava jet',
  robotid: 'BCBCDD45160547239EBF3D8B086CB7F2',
  ip: '10.245.2.120',
  mac: '50:14:79:88:CE:F5',
  sw: 'sanmarino+3.18.11+sanmarino-release-rt321+12',
  sku: 'm611020',
  nc: 0,
  proto: 'mqtt',
  cap:
   { oMode: 4,
     edge: 0,
     maps: 3,
     pmaps: 5,
     tHold: 1,
     tLine: 2,
     area: 1,
     eco: 1,
     multiPass: 2,
     pose: 1,
     team: 1,
     pp: 0,
     lang: 2,
     '5ghz': 1,
     prov: 3,
     sched: 1,
     svcConf: 1,
     ota: 2,
     log: 2,
     langOta: 0,
     tileScan: 1 } }

koalazak, 2021-09-16:
mmm that response is from your other robot (Braava jet). I was looking for a response from the J7.
No worries I saw the correct response in your other issues:
```
{
  "ver": "4",
  "hostname": "iRobot-15DF6D60B10A4BA48A53015E91E330D2",
  "robotname": "Roomba",
  "robotid": "15DF6D60B10A4BA48A53015E91E330D2",
  "ip": "10.245.2.102",
  "mac": "50:14:79:B1:DF:0C",
  "sw": "sapphire+1.0.17+Firmware-Production+52",
  "sku": "j755020",
  "nc": 0,
  "proto": "mqtt",
  "cap": {
    "binFullDetect": 2,
    "dockComm": 1,
    "edge": 0,
    "maps": 3,
    "pmaps": 5,
    "tLine": 2,
    "area": 1,
    "eco": 1,
    "multiPass": 2,
    "pose": 1,
    "team": 1,
    "pp": 0,
    "lang": 2,
    "5ghz": 1,
    "prov": 3,
    "sched": 1,
    "svcConf": 1,
    "ota": 2,
    "log": 2,
    "langOta": 0,
    "expectingUserConf": 1
  },
  "cloudConnState": 14
}
```
I was lookig for the `proto` value.

Looks like a ssl issue. maybe they added more security like client certificates or ssl pinning.

t4sso4, 2021-10-02:
I'm seeing a similar response on an i7 with the latest firmware. Any chance we can extract the password some way?

```
Robot Data:
{
  ver: '3',
  hostname: 'iRobot-3CEE4408011A44209BCE340AFE0A0A23',
  robotname: 'Roomba',
  robotid: '3CEE4408011A44209BCE340AFE0A0A23',
  ip: '192.168.1.204',
  mac: '50:14:79:1C:31:0D',
  sw: 'lewis+3.18.11+lewis-release-rt321+13',
  sku: 'i715020',
  nc: 0,
  proto: 'mqtt',
  cap: {
    binFullDetect: 2,
    oMode: 10,
    dockComm: 1,
    wDevLoc: 2,
    bleDevLoc: 1,
    edge: 0,
    maps: 3,
    pmaps: 5,
    tLine: 2,
    area: 1,
    eco: 1,
    multiPass: 2,
    pose: 1,
    team: 1,
    pp: 0,
    lang: 2,
    hm: 0,
    '5ghz': 1,
    prov: 3,
    sched: 1,
    svcConf: 1,
    ota: 2,
    log: 2,
    langOta: 0,
    tileScan: 1
  },
  blid: '3CEE4408011A44209BCE340AFE0A0A23'
}
Error getting password. Follow the instructions and try again.
```

DiegoCarrilho, 2021-10-02:
I have succesfully extracted password using the IOS app Thor to sniff sll comunication while starting roomba app.
[image]
[image]

t4sso4, 2021-10-03:
@DiegoCarrilho this method worked for me. Had to pay $13 just for that, but it does work pretty well. Looks like the newer firmware has broken the other password options on the i7.

koalazak, 2021-10-03:
J7 owners: can you please enumerate the steps that the iRobot app enumerate to configure your robot the first time?. There is a wizard to setup your robot the first time and there is one step in where the app ask to press some buttons on the robot...which are those butons and instructions?

Can your enumerate the complete guided process?

thank you

twitchdatweek, 2021-10-07:
@koalazak There is no home button on the J7, it is supposed to automatically connect.

Also, is there an alternative to using this "Thor" app? I don't have an iPhone.

koalazak, 2021-10-07:
> @koalazak There is no home button on the J7, it is supposed to automatically connect.
> 
> Also, is there an alternative to using this "Thor" app? I don't have an iPhone.

do you own a J7? Can you please describe the mobile app steps to setup the robot? in detail...

thanks

dcmeglio, 2021-10-08:
From the Roomba website:

This step turns on Roomba®’s internal Wi-Fi network so that your device can locate the robot. This step is not required for Roomba j7. Roomba j7 automatically does this step when turned on initially. **If the robot is unable to be found, please reboot the robot but holding the Clean button until the light ring turns white and spins clockwise (10-20 seconds)**

Maybe try doing that to put the roomba into discovery mode? 

See https://homesupport.irobot.com/s/article/17734

twitchdatweek, 2021-10-08:
> > @koalazak There is no home button on the J7, it is supposed to automatically connect.
> > Also, is there an alternative to using this "Thor" app? I don't have an iPhone.
> 
> Fiddler, Titanium Web Proxy, Charles proxy and …

I tried using Fiddler and have had no success. I have it sniffing traffic when I open the app or reset the roomba and I don't see any messages from iRobot or the IP of the roomba. I'm pretty new to this, so it's probably just me. I even enabled HTTPS.

binary64, 2021-10-12:
I already tried that several ways (holding down the button etc) - but just get `Error getting password. Follow the instructions and try again.`. I get those 7 bytes back and the code for get-roomba-password checks if it's 7 or less, it fails with that error message.

I have just made a discovery, I left my terminal open, and after those 7 bytes.. I started to get traffic! Here's some snippets:

```
:$aws/things/020C465103B5452A92D1E54XXXXXXX/shadow/update{"state":{"reported":{"cleanMissionStatus": {"cycle": "none", "phase": "charge", "expireM": 0, "rechrgM": 0, "error": 0, "notReady": 0, "mssnM": 0, "expireTm": 0, "rechrgTm": 0, "mssnStrtTm": 1633980984, "operatingMode": 0, "initiator": "rmtApp", "nMssn": 41, "missionId": "0001GP94HRQ2A2FJZVJXXXXXXX"}}}}

:$aws/things/020C465103B5452A92D1E54XXXXXXX/shadow/update{"state":{"reported":{"cleanMissionStatus": {"cycle": "clean", "phase": "run", "expireM": 0, "rechrgM": 0, "error": 0, "notReady": 0, "mssnM": 0, "expireTm": 0, "rechrgTm": 0, "mssnStrtTm": 1633980984, "operatingMode": 2, "initiator": "rmtApp", "nMssn": 41, "missionId": "0001GP94HRQ2A2FJZVJXXXXXXX"}}}}

:$aws/things/020C465103B5452A92D1E54XXXXXXX/shadow/update{"state":{"reported":{"cleanMissionStatus": {"cycle": "clean", "phase": "hmPostMsn", "expireM": 0, "rechrgM": 0, "error": 0, "notReady": 0, "mssnM": 0, "expireTm": 0, "rechrgTm": 0, "mssnStrtTm": 1633980984, "operatingMode": 0, "initiator": "rmtApp", "nMssn": 41, "missionId": "0001GP94HRQ2A2FJZVJXXXXXXX"}}}}

:$aws/things/020C465103B5452A92D1E54XXXXXXX/shadow/update{"state":{"reported":{"cleanMissionStatus": {"cycle": "clean", "phase": "evac", "expireM": 0, "rechrgM": 0, "error": 0, "notReady": 0, "mssnM": 0, "expireTm": 0, "rechrgTm": 0, "mssnStrtTm": 1633980984, "operatingMode": 0, "initiator": "rmtApp", "nMssn": 41, "missionId": "0001GP94HRQ2A2FJZVJXXXXXXX"}}}}
```

binary64, 2021-10-12:
My code is just this
```
const tls = require("tls");
var client = tls.connect(
  8883,
  "192.168.1.136",
  {
    rejectUnauthorized: false,
    ciphers: process.env.ROBOT_CIPHERS || "AES128-SHA256",
  },
  function () {
    const magicPacket = new Buffer("f005efcc3b2904", "hex");
    console.log("Connected! Sending magic packet..", magicPacket);
    client.write(magicPacket);
  }
);

client.on("data", function (data) {
  console.log("got data", data, new Buffer(data).toString("hex"));
});
```

HansLehnert, 2021-10-13:
I ran `ROBOT_CIPHERS=TLS_AES_256_GCM_SHA384 get-roomba-password <ip>` and got this output

```
{
  ver: '4',
  hostname: 'iRobot-XXXX',
  robotname: 'XXXX',
  robotid: 'XXXX',
  ip: '192.168.2.128',
  mac: '50:14:79:XX:XX:XX',
  sw: 'sapphire+1.0.17+Firmware-Production+52',
  sku: 'j755020',
  nc: 0,
  proto: 'mqtt',
  cap: {
    binFullDetect: 2,
    dockComm: 1,
    edge: 0,
    maps: 3,
    pmaps: 5,
    tLine: 2,
    area: 1,
    eco: 1,
    multiPass: 2,
    pose: 1,
    team: 1,
    pp: 0,
    lang: 2,
    '5ghz': 1,
    prov: 3,
    sched: 1,
    svcConf: 1,
    ota: 2,
    log: 2,
    langOta: 0,
    expectingUserConf: 1
  },
  cloudConnState: 12,
  blid: 'XXXX'
}
Error getting password. Follow the instructions and try again.
```

After the `f005efcc3b2900` MQTT packet is sent, I get these 13 bytes as response `efbfbd05efbfbdefbfbd3b2903`, which are the same that have been mentioned before. I couldn't make sense of either the request or the response. How was this packet built?

I was able to get the password by capturing traffic from the App and was able to connect with that. For Android 10 I used [apk-mitm](https://github.com/shroudedcode/apk-mitm) to patch the iRobot app and was then able to proxy the requests (through Fiddler, for example).

abhi9cr7, 2021-10-16:
I am using Hoobs and trying to get the blid and password. 

Probably because of Hoobs 4, the folder for hombridge-roomba2 is like this 
`/usr/local/share/.cache/yarn/v6/npm-homebridge-roomba2-1.2.2-20d857384cc8aa25a649f19a66046cc2f25bc718-integrity/node_modules/homebridge-roomba2` and there is no dorita980 within this folder. 

When I did `locate` for dorita980, the directory is `/usr/local/share/.cache/yarn/v6/npm-dorita980-3.1.10-7f6d681688e3fcaa42040b16a9ceded08270457e-integrity/node_modules/dorita980/`

Whenever I am running `sudo npm run getrobotpwd 192.168.XX.XXX` I am getting "can't cd into node_modules/dorita980".

dcmeglio, 2021-11-04:
> Thank you for your support!
>
> In just a couple of hours I have received from you almost half of what is needed to buy a j7 robot
>
> ▰▰▰▰▱▱▱▱▱▱ 41%
>
> I think we are going to make it quicky! thank you!
>
> https://github.com/koalazak/dorita980/blob/master/README.md#donations
>
> thank you to another donator!
> ▰▰▰▰▱▱▱▱▱▱ 47%

How do you set the cipher from rest980?

koalazak, 2021-11-04:
hello @dcmeglio 

```bash
cd path/to/rest980
ROBOT_CIPHERS=TLS_AES_256_GCM_SHA384 npm start
```

koalazak, 2021-11-04:
> Hello folks,
> 
> yeah, looks like the J7 mechanism to get the password is diferent (and will be for next robots too). It is imposible to me debug the problem without a J7 robot. So if you want to help me to get a robot here is the information:
> # Donations to get J7 robot
> 
> https://github.com/koalazak/dorita980/blob/master/README.md#donations
> 
> any other kind of help is welcome!
> 
> thank you very much!

more progress on donations!

▰▰▰▰▰▱▱▱▱▱ 58.4%

thank you guys!

m-ruhl, 2021-11-05:
I tried the workaround/solution from @andornaut, but for me it does not work.

`ROBOT_CIPHERS=TLS_AES_256_GCM_SHA384 NODE_DEBUG=cluster,net,http,fs,tls,timers node run.js`

```
NET 7912: pipe false undefined
NET 7912: _read
NET 7912: _read wait for connection
NET 7912: afterConnect
TLS 7912: start
NET 7912: destroy
NET 7912: close
NET 7912: close handle
NET 7912: _read
NET 7912: _read wait for connection
NET 7912: emit close
NET 7912: pipe false undefined
NET 7912: _read
NET 7912: _read wait for connection
NET 7912: afterConnect
```

Are we sure that the authentication did not change?
i7: `"user_cert": false`
j7: `"user_cert": true`

<details>
  <pre>    "robots":
    {
        "69A56E703ACF491DA483FD225F9F095F":
        {
            "password": "xxx",
            "sku": "i715840",
            "softwareVer": "lewis+3.18.11+lewis-release-rt321+13",
            "name": "Robby deprecated",
            "cap":
            {
                "binFullDetect": 2,
                "dockComm": 1,
                "maps": 3,
                "edge": 0,
                "area": 1,
                "eco": 1,
                "multiPass": 2,
                "pose": 1,
                "pp": 0,
                "5ghz": 1,
                "prov": 3,
                "sched": 1,
                "svcConf": 1,
                "ota": 2,
                "log": 2,
                "team": 1,
                "wDevLoc": 2,
                "bleDevLoc": 1,
                "pmaps": 5,
                "tLine": 2,
                "tileScan": 1,
                "lang": 2,
                "langOta": 0,
                "oMode": 10,
                "hm": 0
            },
            "svcDeplId": "v011",
            "user_cert": false
        },
        "6614293A8BCC403F928CC7BEB11307D4":
        {
            "password": "xxx",
            "sku": "j755840",
            "softwareVer": "sapphire+1.0.20+Firmware-Production+54",
            "name": "Robby",
            "cap":
            {
                "binFullDetect": 2,
                "dockComm": 1,
                "wDevLoc": 2,
                "bleDevLoc": 1,
                "edge": 0,
                "maps": 3,
                "pmaps": 5,
                "tLine": 2,
                "area": 1,
                "eco": 1,
                "multiPass": 2,
                "pose": 1,
                "team": 1,
                "pp": 0,
                "lang": 2,
                "5ghz": 1,
                "prov": 3,
                "sched": 1,
                "svcConf": 1,
                "ota": 2,
                "log": 2,
                "langOta": 0,
                "expectingUserConf": 1
            },
            "svcDeplId": "v011",
            "user_cert": true
        }</pre>
</details>

dcmeglio, 2021-11-05:
Worked for me. What version of node do you have? It requires TLS 1.3 which I believe means you need at least node 12

m-ruhl, 2021-11-05:
@dcmeglio oh thanks 👍  did not had that in mind.. there was still node 10 in use

StefCGN, 2021-11-11:
Seems, when the robot is new, it needs a little bit more time.
After creating the Map on the phone and several "discovery-runs", i was able to connect the i7 again.
Perhaps this also helps with some j7?

Danny2100, 2021-11-14:
Has anyone figured out how to do this on a non rooted Android device. It appears impossible.

dcmeglio, 2021-11-14:
Do you have to root an Android device to do this??? On iOS you can just download an app such as Thor or Charles from the app store. There is no equivalent for Android?

Danny2100, 2021-11-14:
> Do you have to root an Android device to do this??? On iOS you can just download an app such as Thor or Charles from the app store. There is no equivalent for Android?

No it needs to be rooted. Surprisingly Android is more secure in the sense that it does not allow a user to install a self-signed certificate and then use it. Apps also can not import certificates or generate additional certificates. You can install a certificate but a sniffing app for example can't use it if I'm not mistaken. I tried a bunch of ways. The apps have no access to device certificates except for their own and you would need to edit the app somehow from what I read to tell it where to find the self-signed certificate.

dcmeglio, 2021-11-15:
Oh wow. I didn't know that. So another option would be to run something like Fiddler on your PC and then set it as an HTTP proxy on your android device. I don't own an Android device so I have no idea how accurate these instructions are, but https://www.digitalcitizen.life/how-add-proxy-server-wireless-connection-android/

Danny2100, 2021-11-15:
I will try that my other fear is Roomba's app may not even work because of
it's authentication methods. I had fiddler installed but I didn't do the
proxy I will do that later this week and report back.

dcmeglio, 2021-11-15:
I haven't tried Fiddler with Roomba before, but I've used it successfully with many other apps and it _usually_ works.

uvNikita, 2021-11-15:
No need to root Android to get the password with proxy. You just have to modify iRobot apk using this tool to enable user certificates: https://github.com/shroudedcode/apk-mitm.

mjg59, 2021-12-03:
As far as I can tell, there's no longer a mechanism to get the password via the local API other than during initial device setup:

```
    public String getPassword() {
        Assert.m4257g("Password is not provided to local secure socket network session.", false);
        return null;
    }
```

instead, it's necessary to retrieve it from the cloud service. This is a multi-stage process. Initially a POST is made to https://accounts.us1.gigya.com/accounts.login including an API key and the username/password. This returns an oauth2 token. A further POST is then made to https://accounts.us1.gigya.com/accounts.getAccountInfo which includes the oauth token, a nonce, a timestamp and a signature generated using a secret. This returns a bunch of json including UID, UIDSignature and signatureTimestamp fields, and a final POST is made to https://unauth2.prod.iot.irobotapi.com/v2/login which includes that as part of a JSON payload. This finally returns another JSON blob that includes information about the vacuums.

https://github.com/SAP/gigya-python-sdk should (theoretically) be sufficient for generating the signatures, but it still requires knowledge of the secret, and I haven't been able to extract that from the app yet. The API key is 3_rWtvxmUKwgOzu3AUPTMLnM46lj-LxURGflmu5PcE_sGptTbD-wMeshVbLvYpq01K.

mjg59, 2021-12-03:
Ah, I think I've got it. Submit to https://accounts.us1.gigya.com/accounts.login with the following parameters as form encoded parameters:

```
loginID=username
password=password
apiKey=3_rWtvxmUKwgOzu3AUPTMLnM46lj-LxURGflmu5PcE_sGptTbD-wMeshVbLvYpq01K
targetEnv=mobile
sessionExpiration=-2
```

This will return a json blob with a whole bunch of data, including a sessionInfo object that looks something like:

```
  "sessionInfo": {
    "sessionToken": "st2.s.AcbHR0jG2g.kG_mr4g1reIyuOK2B30nxkotlJE2MJx2m30vf3lFfyBDreY3v9ryBPhjlW3Q_pYA9gtHa0jgokucYtTr7hM7G3-XK5AgQGCjUOFgMi63tlQ.QrMEhIvS-qkad4nhnurMDj4tGJZmkL9NiOJseOaOBSc3om-uHt1BjeGgkpMgUlASKEj0KwtC9-GjzHwq7iTydg.sc3",
    "sessionSecret": "WMEqGqQfqCPe3lt6zFDM0cvbB2s="
  }
```

Then POST to https://accounts.us1.gigya.com/accounts.getAccountInfo, again using form encoded parameters:

```
nonce: (a random string, the implementation is the current unix time in milliseconds followed by a _ followed by a random integer)
timestamp: (the unix timestamp)
oauth_token: (the sessionToken from the previous step)
targetEnv: mobile
sig: (the calculated signature, see https://github.com/scotthovestadt/gigya/blob/master/lib/sig-utils.ts, using the sessionSecret from the previous step)
```
Obtain the UID, UIDSignature and signatureTimestamp from the response to this, then POST a JSON object to https://unauth2.prod.iot.irobotapi.com/v2/login containing:

```
{
    "app_id": "ANDROID-C7FB240E-DF34-42D7-AE4E-A8C17079A294",
    "assume_robot_ownership": "0",
    "gigya": {
        "signature": (UIDSignature),
        "timestamp": (signatureTimestamp),
        "uid": (UID)
    }
}
```

and this will give back JSON that contains robot data including passwords.

mjg59, 2021-12-03:
Looking into https://github.com/scotthovestadt/gigya/, it looks like this is just a matter of:

```
const gigya = new Gigya('YOUR_API_KEY', 'YOUR_DATA_CENTER', 'YOUR_USER_KEY', 'YOUR_USER_SECRET');
```

where data center is US1, API key is 3_rWtvxmUKwgOzu3AUPTMLnM46lj-LxURGflmu5PcE_sGptTbD-wMeshVbLvYpq01K, and then the user creds. You should then be able to call login and then gigya.accounts.getAccountInfo to get the data including the UID, signature and timestamp, then send that to https://unauth2.prod.iot.irobotapi.com/v2/login.

Danny2100, 2021-12-03:
Is there a script for this? How would someone without as much knowledge as you pull this off? Instructions maybe?

mjg59, 2021-12-05:
Ah, it's actually easier than I thought - the gigya account info isn't necessary. Replace USERNAME and PASSWORD in this script and it should get 

```
#!/usr/bin/env python3                                                          

import random
import requests
import time

apikey = "3_rWtvxmUKwgOzu3AUPTMLnM46lj-LxURGflmu5PcE_sGptTbD-wMeshVbLvYpq01K"

data = {"apiKey": apikey,
        "targetenv": "mobile",
        "loginID": USERNAME,
        "password": PASSWORD,
        "format": "json",
        "targetEnv": "mobile",
        }

r = requests.post("https://accounts.us1.gigya.com/accounts.login", data=data)

response = r.json()

data = {"timestamp": int(time.time()),
        "nonce": "%d_%d" % (int(time.time()), random.randint(0, 2147483647)),
        "oauth_token": response['sessionInfo']['sessionToken'],
        "targetEnv": "mobile"}

uid = response['UID']
uidSig = response['UIDSignature']
sigTime = response['signatureTimestamp']

data = {
    "app_id": "ANDROID-C7FB240E-DF34-42D7-AE4E-A8C17079A294",
    "assume_robot_ownership": "0",
    "gigya": {
        "signature": uidSig,
        "timestamp": sigTime,
        "uid": uid,
    }
}

r = requests.post("https://unauth2.prod.iot.irobotapi.com/v2/login", json=data)

response = r.json()
for robot in response['robots']:
    print("BLID:", robot, "Password:", response['robots'][robot]['password'])
```

koalazak, 2021-12-05:
@mjg59 man you rock!

published `dorita980@3.1.11` supporting this new method. Readme updated too.

```
$ npm install -g dorita980
$ get-roomba-password-cloud myemail@example.com myeasypassword
Found 1 robot(s)!
Robot "Dorita" (sku: R98---- SoftwareVer: v2.4.16-126):
BLID=> xxxxxxxxxxxxx
Password=> :1:1486937829:gktkDoYpWaDxCfGh <= Yes, all this string.

Use this credentials in dorita980 lib :)
```

I will fine tune the code in next version.

Thanks to all the donors! I bought the j7 on black friday and it's on its way! It will help to implement all the cloud API.

thank you @mjg59 !

closed 2021-12-05 by koalazak

mjg59, 2021-12-05:
I did some more digging - the "best" first step is to hit:

https://disc-prod.iot.irobotapi.com/v1/discover/endpoints?country_code=US

with the appropriate country code. This will provide the API Key, the domain to use for gigza, the host to use for later auth and also the host to use for communicating with cloud services. This avoids hardcoding things that may change over time.

koalazak, 2021-12-05:
yes! that's what I meant by "fine tune the code", also generating a random app_id and options to use ENV vars to set configurations.

thank you!

mjg59, 2021-12-05:
Not strictly related, but https://github.com/mjg59/python-irobotapi also gives an example of using the AWS-based IoT API to retrieve mapping room data, and should be easily extensible to other API calls.

rahmanonik18, 2023-06-12:
https://disc-prod.iot.irobotapi.com/v1/discover/endpoints?country_code=US

How do you retrieve this URL? Is it mentioned in the official documents of the app? How can i retrieve this url?

creeryk, 2023-08-24:
This is what finally worked for me:
sudo docker run -it node sh -c "npm install -g dorita980 && get-roomba-password-cloud myemail@example.com myeasypassword"