koalazak/dorita980 #145: How to Determine if Robot State is Paused?

rcoletti116, 2021-09-20:
I've been looking through the code, docs, and old issues trying to determine how we can tell if the robot is in a paused state. I've not seen anything to this effect in getMission or getRobotState. Does anyone know?

I've seen reference to bbpause with various numerical values. Does anyone know what these mean?

rcoletti116, 2021-09-21:
For those who might be following I've got two methods I'm exploring.

1. Keying off 'lastCommand', e.g., 

lastCommand: {
command: "pause",
time: 1632170499,
initiator: "manual"
},

OR

2. cycle + phase combo.  Cycle is "clean" and the phase is "stop" when it is paused while cleaning, while the cycle property is "none" when it is fully stopped, not engaged in a job.

closed 2021-09-21 by rcoletti116

jeremywillans, 2021-09-22:
Hey @rcoletti116, I mapped out a lot of the settings in my HA Package for rest980/dorita980.

This should give some guidance for this and other options - ref [here](https://github.com/jeremywillans/ha-rest980-roomba/blob/master/vacuum.yaml#L81)

rcoletti116, 2021-09-22:
Thanks @jeremywillans!