koalazak/dorita980 #148: v3.20.7 position issue

crissmil, 2021-11-08:
Roomba S9+ after v3.20.7 no "pose" data available

crissmil, 2021-11-11:
Today also my Braava m6 updated to v3.20.7 and "pose" data disappeared. Any ideas?

StefCGN, 2021-11-13:
Same here with i7

warki, 2021-11-14:
Same here. Tried to read the MQTT-Data, but the position is not updated to the MQTT server. 

Did anyone already recorded the cloud dialog between the App and iRobots Server? Maybe there is the data. And maybe even better a routing from the current roomba position coordinates to the current room id or name (as seen in the App).

NickWaterton, 2021-11-30:
We were reading the aws shadow updates (which we still see), but in the latest version of firmware, iRobot seems to have stopped sending “pose” updates to the aws shadow.

not sure what we can do about that.

tobuh, 2021-12-05:
might it be a good idea that we all bother iRobot, why they removed such a feature?

koalazak, 2021-12-05:
yes, looks like the new firmwares are not reporting the position anymore :/

NickWaterton, 2021-12-05:
Might be something that can be turned on or off, we just don’t know the command to turn it on.

henripalmroth, 2021-12-08:
Just had a chat with support:

[image]

brad07x, 2021-12-24:
Has anyone made progress on this issue with their own investigation or with iRobot support? 

I was setting up rest980/HA for my Roomba s9+ and Braava m6 - both are running 3.20.7 and no 'pose' updates.

Thanks!

henripalmroth, 2022-01-13:
Not looking good from support perspective:

[image]

pahenning, 2022-01-14:
Try to think like these people: You would never take such a feature out of the software, but rather hide it such that it must be switched on. And then you would try to hide the command for activating the position messages as long as possible. So lets have a look.

In the messages received through the mission, two suspicious settings appear:
```
{"state": {"reported": {
....
  "featureFlags": {
        "quietNav": true,
        "homeMonitoring": false,
        "clearHaz": true,
        "hibLed": 0,
        "covPlan": 0,
        "ros2SptLvl": true
    },
...
 "smartHome": {"homeMonitoringAllowed": false},
...
}
```
Unfortunately, so far I was unable to modify these settings. At least, the usual "delta"-command fails - none of these
```
  delta {"state": {"homeMonitoringAllowed": true}}
  delta {"state": {"featureFlags": { "quietNav": false}}}
  delta {"state": {"featureFlags": { "homeMonitoring": true}}}
```
works. Maybe someone has another idea?

Regards

pah

NickWaterton, 2022-01-14:
This is what I was thinking as well.

We may have to do more experimenting, to see if we can change these settings. `quietNav` looks to be a good target to start with.

pahenning, 2022-01-14:
That is exactly what I did, as you see above. The delta topic works when setting such things as OpenOnly.
I should add that I am not using dorita, but have written other software running in the home automation system FHEM (http://www.fhem.de). I even had this software producing nice SVG maps of the cleaning runs.
But this irrelevant as long as we can join forces to crack this problem here.

Regards

pah

NickWaterton, 2022-01-14:
I have triedthe same thing, but with no luck. I also note that my M6 mop reports:
            "featureFlags": {
                "quietNav": false,
                "clearHaz": true,
                "hibLed": 0,
                "covPlan": 0,
                "ros2SptLvl": true
            }
And it does not report the pose location either (since the last update). So `quitetNav` does not seem to be it.

Previously this would be reported:
            "cap": {
                "pose": 1,
                .
                .
            }
But now `"pose"` is missing from the `"cap"` fields, and `"pose"` is what was reporting the position of the roomba.

This may be more difficult than we were hoping.

mehalter, 2022-01-14:
I'm guessing `cap` is short for capabilities. Maybe adding `"pose": 1` back to the `cap` field would trick it into reenabling since `quietNav` didn't work

NickWaterton, 2022-01-14:
Tried that, and it didn't work either.

pahenning, 2022-01-14:
I bet these guys see themselves as customer-oriented...
The best way to proceed here would be a complete sniff of the data sent to the cloud during a short cleaning run.

Regards 
pah

NickWaterton, 2022-01-14:
I have a decoded capture of the app connecting to the cloud, and starting up. I haven't captured an entire run. I haven't been able to find much in there (other than how to connect to AWS, authenticate and retrieve BLD and password, maps etc via REST). I can see all the MQTT messages that the robot sends to the cloud, but again, not getting much either.

I suspect that iRobot just doesn't care about what we are doing, they don't use the tracking info, so they removed it (might have been something they used in development, but don't anymore). Might have been for a "real time maps" feature that they never implemented. I'm hoping they didn't remove it completely, just disabled it, or report it a different way. We may be out of luck though.

The robot must report some sort of tracking data though, as the app does tell you what room the robot is currently in when it's running. As you say a capture of a full run may be useful. Having said that, if the app uses an MQTT connection to AWS, I can't capture it with my current tools (which is annoying).

The other thing I was considering is a decompile of the android app. I have an android VM that I can spin up for these kinds of things, and sometimes you can extract useful data that way.

This may take some time...

mehalter, 2022-01-14:
It seems like the x/y coordinates have to be accessible somewhere in the communication since it generates an image in the app of what parts were cleaned

NickWaterton, 2022-01-14:
Yes, but the map could be built on the robot itself. Previously the map may have been built in the cloud due to limited resources on the robot. It could be that a raw map is uploaded to the cloud at the end of a run. We just don't know at the moment how its done.

pahenning, 2022-01-15:
A hint towards this latter suspicion: I own three robots, 960, 981 and i7. The older models still tell their position, and from a run of the 960 I get some 3500 points, while from an i7 in roughly the same area I (had) received only about 300 points.

Regards
pah

koalazak, 2022-01-16:
im with @NickWaterton : position is not longer part of the exposed state because they (irobot) don't need that to show you the map.
If they put a flag to enable/disable this: we can change it with setPreferences() to edit the state which uses the delta command:

```javascript
myRobotViaLocal.setPreferences({mapUploadAllowed: false})
```

did you test with tihs?

```
myRobotViaLocal.setPreferences({featureFlags: {pose: true}});
myRobotViaLocal.setPreferences({featureFlags: {position: true}});
myRobotViaLocal.setPreferences({featureFlags: {enablePose: true}});
myRobotViaLocal.setPreferences({featureFlags: {enablePosition: true}});
myRobotViaLocal.setPreferences({featureFlags: {enablePose: true}});
myRobotViaLocal.setPreferences({pose: true});
myRobotViaLocal.setPreferences({enablePose: true});
myRobotViaLocal.setPreferences({enablePosition: true});
myRobotViaLocal.setPreferences({position: true});
myRobotViaLocal.setPreferences({allowPose: true});
myRobotViaLocal.setPreferences({allowPosition: true});
myRobotViaLocal.setPreferences({poseAllowed: true});
myRobotViaLocal.setPreferences({positionAllowed: true});
// etc
```

pahenning, 2022-01-29:
I tried all these:

  ' delta {"state": {"homeMonitoringAllowed": true}}';
  ' delta {"state": {"featureFlags": { "quietNav": false}}}';
  ' delta {"state": {"featureFlags": { "homeMonitoring": true}}}';
  ' delta {"state": {"featureFlags": {"pose": true}}}';
  ' delta {"state": {"featureFlags": {"position": true}}}';
  ' delta {"state": {"featureFlags": {"enablePosition": true}}}';
  ' delta {"state": {"featureFlags": {"enablePose": true}}}';
  ' delta {"state": {"pose": true}}';
  ' delta {"state": {"enablePose": true}}';
  ' delta {"state": {"enablePosition": true}}';
  ' delta {"state": {"position": true}}';
  ' delta {"state": {"allowPose": true}}';
  ' delta {"state": {"allowPosition": true}}';
  ' delta {"state": {"poseAllowed": true}}';
  ' delta {"state": {"positionAllowed": true}}';
  
  so far no effect.
  
  Regards 
  pah

drizzlerR, 2022-03-03:
Hello,
with my Roomba 980 (firmware 2.4.16-126) there are also no pose information anymore.

Regarding the quietNav entry : iRobot introduced a room traveling feature that shuts down the main suction unit and the rollers while traveling between cleaning areas. It is named "quiet drive". The flaq should be disabled or non existend with older 9xx models.

rpitera, 2022-04-14:
Glad I found this; was thinking it was something in my configs but now I can see this due to the firmware change.  And here I was happy that my i3 had gained the ability to distinguish rooms after a mapping run; didn't realize I was losing my HA maps.  Anyway, I wanted to add the i3 (3150) to the list of affected models and make sure I got any updates on progress here.  Happy to help test if I can or provide any information.

generalk25, 2022-04-21:
Mine is broken as well, I7+

Angel0ffDeath, 2022-04-25:
It seems there is a new firmware:
 https://homesupport.irobot.com/s/article/26887
released Apr. 20th, 2022, but my s9 still not receive it.
Let's hope they fixed this positioning issue

rpitera, 2022-04-25:
@Angel0ffDeath - From your lips to God's ears. my friend.  Thanks for the heads up.  I think I'll try to stay in the 'control group' just in case it locks off something else though.

Angel0ffDeath, 2022-04-26:
Updated to the new firmware 22.7.2, but situation is the same - no positioning data. I think they totally changed concept of data transmission. I created new map - it takes few minutes to be uploaded to the cloud, which means the map is created in Roomba. Mqtt explorer - as in previous firmware only rssi is updated each second. The rest data is updated about each 10 min... probably they want to reduce cloud traffic. On first sight the only new function is child/pet lock - lock buttons on roomba. In other words (according to me) if someone still wants live positioning update - find a way to revert back to older firmware and block cloud access....

rpitera, 2022-04-26:
Damn. Was hoping for some better news than that. AFAIK, the only way to rollback to a previous firmware (without a reset of course) is to contact iRobot support.  And I doubt they'd trip a rollback for you on the basis of the positioning data problems.  I'm still keeping hope knowing better minds than me are thinking about the same problem.

I'd try a reset but I'm afraid I'd end up losing what I still have - control and status - which are my main concerns.  I like having the map, but honestly I could live without it and still be happy with what @koalazak has given me.  Thanks for the info.

Angel0ffDeath, 2022-04-26:
@rpitera  Personally for me actually I dont see any precious info which live tracking will give me. It is enough to know robot is cleaning and how exactly - well it is his problem. Moreover I actually dont have time to supervise the robot (thats the reason I have it). The rest is just to show your friends how clever/smart is your vacuum cleaner to your friends, which also makes no sense...
It is still possible to remotely run the robot with certain map and to forget it... Why do you need to know everything???
I think what I get from robot is enough for me and will stop investigating and searching solutions during the last almost 6  months -there are more important things to do
@koalazak I suggest you to close this issue and to cleanup the code...

pawelpp85, 2022-05-02:
Is there a way to figure out in which room the robot is currently in? I would like to use HA to turn on the lights for my i7 and m6.

SageKaz, 2022-05-27:
I'm wondering if anyone has tried setting the `maps` value under cap.

`
"cap": {
	"pose": 1,
	"maps": 1,
	...
}
`
I still have pose capabilities on my model so I can't test it out, but I've seen other values mentioned, but not specifically the `maps` value yet.

Mirarkitty, 2022-09-24:
No news I suppose. Anything from support?

Wolk9, 2022-10-29:
Anyone any news on this?

ThellraAK, 2022-12-12:
@NickWaterton I can do a full packet capture from reboot to end of cleaning cycle, but I wouldn't know how to start doing anything with it, is that something that would help you?

MattePozzy, 2024-05-10:
I suppose there isn't any news, right? :(

mitchcapper, 2025-01-21:
@maa-x  the paper talks about getting map data from the cloud, and (if my brief look at the source: https://github.com/BiTLab-BaggiliTruthLab/PyRoomba is correct) historical map data only.   This bug is about the local access to map data and live positioning.

NickWaterton, 2025-01-21:
@mitchcapper 
Agreed, but this does show that the Roomba is storing all the co-ordinates - the question is - does it upload them all in one go at the end of the mission, or is it continually updating via some unseen method?

The app *does* report which room the Roomba is currently cleaning, so there is some real time data transfer.

It would be interesting to run this while the Roomba is vacuuming, and see when the map is created.

maa-x, 2025-01-21:
Ah my bad I hadn't realised this issue was specifically about local map data. As far as I'm aware, beyond that repo there have been no mentions of it being possible without Roomba's application, cloud or local. 

@NickWaterton, I've tested it while the Roomba is vacuuming and did not get a live map nor did the mission history update until it was done. 

Here's the path they query to get the map data: `https://auth3.prod.iot.irobotapi.com/v1/robotid/pmaps/pmap_id/versions/pmapv_id/.` Though it doesn't look like we could translate that into a hidden MQTT topic ..

mitchcapper, 2025-01-22:
> Agreed, but this does show that the Roomba is storing all the co-ordinates

Does it?  We know we can build the map, and we know we have coordinates for events, but I am not sure we get the trace style output before where you could reconstruct the path through rooms.

I did MITM some of the traffic for the pmaps/pmap_id/* urls and didn't see anything that looked like a full path dump.  Now I only passively monitored and didn't bother trying to construct my own requests and do the signing for them so its possible.

I did also look at their code and there are the coordinates for the map fetch but that is not a per run item.

I think the two items this repo focused on around this issue  were  live position data or accessing historical data we couldn't before like path taken but sadly I don't think we have either of those here.

maa-x, 2025-01-22:
@mitchcapper , I'm unsure if it's stored on the cloud or on the robot itself, but see the screenshots below for the path data from a few different runs:

[image]

[image]

[image]

mitchcapper, 2025-01-23:
OK I found some issues with the original library and have made some improvements.  If you have multiple robots it picked the first which in my case was a nearly unused robot.  I have exposed the entire timeline as well for the runs.  You can find the updates at https://github.com/mitchcapper/PyRoomba

Sadly, as mentioned, there is not path data stored in what I see.  In your photos the "cleaned area" fills look a bit more line like, probably because of a smaller total map size, but it is really just a set of shapes and they are not timestamped.

You essentially have layers for most of the generic stuff and they are just polygons:

[image]

Now the initial mission data does include timestamped events and a few of those might be useful to kind of create a map but it isn't a detailed path as they don't have coordinates.  Im guessing best is what room did it go between:

[image]

My favorite in there is kidnap.  I'm really hoping that is their name for when you pick the unit up.   Essentially only the travel/room events give you location data (travel tells you target room and room tells you the room it just cleaned).

Anyway  I am not sure if I will touch it much more any time soon as the more interesting data to me is still not really there.