koalazak/dorita980 #153: Progress on j7+

farberm, 2021-12-08:
I have obtained the blid and password with dorita980... but I cannot connect to the j7+ by using the homebridge-roomba2 plug-in.  Have you made any progress after your new purchase with any updates?

Thanks in advance.

Danny2100, 2021-12-11:
I'm using home assistant and using the container to run dorito980. I got the blid and pw. How do I run the ```myRobot.getRobotState(waitForFields)``` as in what's the command that needs to coincide with that command when I enter it into the command line.

bloomkd46, 2022-01-04:
I have someone with the same problem using [homebridge-iRobot issue#2](https://github.com/bloomkd46/homebridge-iRobot/issues/2) it immediately emits offline whenever you try to connect

Danny2100, 2022-01-04:
I tried using rest980 but all I get is {} when I attempt to load the state and I know it's the right address as it actually has a Roomba logo and the addon in the logs shows the state was requested.

CoreyJ87, 2022-01-09:
> I tried using rest980 but all I get is {} when I attempt to load the state and I know it's the right address as it actually has a Roomba logo and the addon in the logs shows the state was requested.

Do you have the official roomba integration installed? I had to disable that. They conflict

Danny2100, 2022-01-09:
> > I tried using rest980 but all I get is {} when I attempt to load the state and I know it's the right address as it actually has a Roomba logo and the addon in the logs shows the state was requested.
> 
> Do you have the official roomba integration installed? I had to disable that. They conflict

It's disabled

koalazak, 2022-01-16:
the state maybe empty {} in the first event and then start filling with data in subsecuent events

Danny2100, 2022-01-16:
> the state maybe empty {} in the first event and then start filling with data in subsecuent events

For me the issue was the rest980 addon. It was updated now with support for the j7 and it works now.

koalazak, 2022-01-16:
perfect

closed 2022-01-16 by koalazak