koalazak/dorita980 #157: j7+ with version 4 not connecting

ricardopineda93, 2022-02-23:
Just got the j7+ and have been following the docs closely, however running Local and attempting to connect to my Roomba just hangs until I kill. Running `nc -zv <roomba ip> 8883` rejects the connection while my script is running and hanging so I am assuming something is happening but not really resolving? 

MacOS Monterey 12.2.1
Node 14.8.1
Dorita980 3.1.11
Roomba J7+ 

My test script:
`const dorita980 = require("dorita980");
const { blid, password, ip } = require("./config.json");

const roombaLocal = new dorita980.Local(blid, password, ip, 2);

roombaLocal.on("connect", () => {
  roombaLocal
    .getSys()
    .then((sys) => {
      console.log(sys);
      roombaLocal.end();
    })
    .catch((err) => console.error(err));
});
`

Running the discovery command yields:

`{
  ver: '4',
  hostname: 'iRobot-XXXX',
  robotname: 'XXXX',
  robotid: 'XXX',
  ip: '192.168.1.8',
  mac: 'XXXX',
  sw: 'sapphire+1.2.12+Firmware-Production+57',
  sku: 'XXXX',
  nc: 0,
  proto: 'mqtt',
  cap: {
    binFullDetect: 2,
    oMode: 2,
    odoa: 2,
    dockComm: 1,
    edge: 0,
    maps: 3,
    pmaps: 5,
    tLine: 2,
    area: 1,
    eco: 1,
    multiPass: 2,
    team: 1,
    pp: 0,
    lang: 2,
    '5ghz': 1,
    prov: 3,
    sched: 1,
    svcConf: 1,
    ota: 2,
    log: 2,
    langOta: 0,
    expectingUserConf: 1
  },
  cloudConnState: 14
}`

iRobot app says "Robot Software 1.2.12 Updated February 10, 2022".

nriley, 2022-03-21:
I'm getting this too with a j7. Is this working for anyone else? I have no trouble getting the BLID/password from the web service but nothing local seems to connect, even after I tried rebooting the vacuum.

fieldsjm, 2022-03-24:
Same here, successfully retrieved BLID/password but timesout when attempt a GET request for data

rcjkierkels, 2022-04-26:
Love to see this working for the j7 :-)

jfroy, 2022-05-19:
Same problem, with firmware 22.7.2-2022-03-29.

mikesalz, 2022-05-23:
I am excited to get this working, but am experiencing the same issue.  @koalazak Just curious if you have any thoughts?

awawis, 2022-07-13:
Same issue here. Curious to know is j7 support is in the works.

julianpoy, 2022-08-16:
Experiencing the same. @koalazak do you know where I'd start to look into debugging this?

mikesalz, 2022-08-16:
@koalazak doesn't seem to reply to inquiries for some reason.  This might be helpful...
https://community.home-assistant.io/t/roomba-j7-integration-help/339705/28

sjorge, 2023-05-15:
Also seeing this with a J7+, I tried the cipher change mentioned but that didn't help. Neither can I say curl the endpoint dorita980 is trying to access.

vladimir-aubrecht, 2023-06-03:
I played with it today and I have some discoveries, hopefully somebody will manage to get closer than me.

Dorita client is managing to get Roomba J7+ into some weird state which causes, that nothing can connect anymore after some amount of tries. After Roomba J7+ reboot everything works normally; I'll clarify that.

I managed to connect with [MQTT Explorer](http://mqtt-explorer.com) app for MacOS with settings:
- Protocol: mqtt://
- Host: my roomba ip
- Port: 8883
- Validate Certificate: Disabled
- Encryption (tls): Enabled
- Username: my roomba username obtained through the script
- Password: my roomba password obtained through the script
- MQTT Client ID (in Advanced - must be set!): my roomba username obtained through the script

With these settings, after reboot of roomba I am able to connect and I can see messages from roomba about battery, signal strength, etc. So I am sure, it's possible to connect there 😊 

Unfortunately did not have same luck with Dorita client library where after vacuum reboot I am getting:
```
Error: connect ECONNREFUSED 192.168.2.13:8883
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1571:16) {
  errno: -61,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.2.13',
  port: 8883
}
```

Did not find what is error -61. One way which should uncover how to fix it could be using some network sniffer tool like Wireshark or something else. It's possible to capture traffic from MQTT Explorer, then capture traffic from Dorita library and compare them to find out differences and reasons why it works from one, but not another. Thou it's almost 6 am here now, so it's a job for another day ... 😊 (actually probably another week/month, I'll not have time to continue on it soon 😕 ).

Zahlii, 2023-06-22:
The problem most likely is usage of node after v16, which leads to an internal error "unsafe legacy renegotiation disabled" which you don'rt control on client side. Make sure you are using node 16xxx

vladimir-aubrecht, 2023-07-01:
Thank you for suggestion, but unfortunately it seems there is some other problem there. I tried it in two ways:
- Through NodeRed plugin using Dorita980 library ([node-red-contrib-roomba980-fw2](https://github.com/yuchangyuan/node-red-contrib-roomba980-fw2)).
- And directly by using Dorita980 library with the script below. In script I never receive "connect event".

All testing were performed with Node v16.16.0.

Script which I am using for testing:
```js
var dorita980 = require('dorita980');
console.log("Creating connection");
var myRobotViaLocal = new dorita980.Local('myusername', 'mypassword', 'roomba ip'); // robot IP address

console.log("Registering connect handler");
myRobotViaLocal.on('connect', init);
console.log("Handler registered");

function init () {
    console.log("connected");
}
```

[image]

[image]

koalazak, 2023-07-07:
hey guys, sorry im late (?) haha

The problem is that[ I don't have a j7+ robot](https://github.com/koalazak/dorita980#donations) to test and troubleshoot this connection issue. You might need to play around with the connection options and certificates. I saw that you tried to upgrade mqtt from 2 to 4, keep in mind that there are [breaking changes](https://github.com/mqttjs/MQTT.js#notes) in error handling. Also, try to maintain compatibility with older versions of node.js and old robots if you are going to send PRs. 
Thank you!

CarComp, 2024-03-25:
@koalazak It is happening now on the 980. Can you confirm?

koalazak, 2024-03-25:
it is NOT happening on my 980.

tested and working with node version 10, 14 and 20.

robot firmware and discovery info:

```
{
  ver: '3',
  hostname: 'Roomba-xxxxxxxxxxxx',
  robotname: 'xxxxxx',
  ip: 'x.x.x.x',
  mac: 'xx:xx:xx:xx:xx',
  sw: 'v2.4.17-138',
  sku: 'R98----',
  nc: 1,
  proto: 'mqtt',
  cap: {
    pose: 1,
    ota: 2,
    multiPass: 2,
    carpetBoost: 1,
    pp: 1,
    binFullDetect: 1,
    langOta: 1,
    maps: 1,
    edge: 1,
    eco: 1,
    svcConf: 1
  }
}
```

Mobile App say that firware was upgraded on 13 oct 2023 and say there is no new firmware.

Remember the reobot allows only 1 connection at the time (rejecting new connections)

closed 2024-03-25 by koalazak

CarComp, 2024-03-25:
@koalazak I get that part works perfectly... try the "get schedule" command.

koalazak, 2024-03-25:
that is what I tested and works for me. I am able to connect to the robot and receive the state and send commands.
the discovery info was just to show my "setup"