koalazak/dorita980 #162: Found 0 robot(s)!

FredLiu0512, 2022-07-28:
I have added them either in the iRobot mobile app or on the official website, but still can't find any robots.
What should I do next?
Where exactly should I add robots？
Please help me. Thank you.

Utopiah, 2022-08-08:
Did you add the robot to the same WiFi network as the computer you are trying to use dorita980 on? If not make sure to do so first. After that, the robot will say so and display a WiFi symbol on it, use `nmap` to scan for devices and you should see a device from `AzureWave Technology`. If not then you are probably on different networks. Also make sure to disconnect from the app as only 1 connection is possible. You can turn WiFi off from your phone to be sure.

heiybb, 2022-09-12:
> I have added them either in the iRobot mobile app or on the official website, but still can't find any robots. What should I do next? Where exactly should I add robots？ Please help me. Thank you.

Are you using the iRobot China Mainland version?
If so, seems that Chinese version firmware have a different configuration, and not compatible with repository.

FredLiu0512, 2022-10-11:
Thanks. I have figured it out, and it’s because of the button error. Fabien Benetou ***@***.*** ***@***.***)> 2022年8月8日 16:29:16 wrote: > > > Did you add the robot to the same WiFi network as the computer you are trying to use dorita980 on? If not make sure to do so first. After that, the robot will say so and display a WiFi symbol on it, use nmap to scan for devices and you should see a device from AzureWave Technology. If not then you are probably on different networks. Also make sure to disconnect from the app as only 1 connection is possible. You can turn WiFi off from your phone to be sure. > > > > — >

closed 2023-07-07 by koalazak