koalazak/dorita980 #163: Change of owner (Amazon aquisition) and pinning firmware

Utopiah, 2022-08-07:
Hi, I used the library in the past and it was quite fun. I mostly tinkered with it to integrate with my WebThings IoT gateway. Now though that I see Amazon acquired iRobot my consideration is very different.

I'd like to skip the app entirely, make sure that no data is sent back to Amazon servers (regardless of what information that is) but still be able to use the device.

My question simply is : does this sound like a reasonable plan knowing that in 2020 (see https://twitter.com/utopiah/status/1322456803603582976 , times flies) when I tinkered with it, dorita980 did everything I hope for?

Will I still need to do an update before `get-roomba-password-cloud`? If not, does it ever expire?

PS: I'd rather flash with an open-source firmware but I'm not aware of any.

Utopiah, 2022-08-08:
First little hiccup, I saved the MAC address of the device and `nmap` couldn't find it. I had to install the app to fix the WiFi.

I imagine next time I would change the WiFi I would have to use `wlcfg` with `setPreferences()` before disconnecting it. Otherwise I would have to rely on the app which would in turn probably force to agree to a new ToS and probably push for firmware updates.

Utopiah, 2022-08-08:
Could setting up `cloudEnv` or another configuration parameter to a specific value prevent from attempt to query home? It would be more convenient that trying to block at the (local, e.g PiHole) DNS level (assuming it relies on it rather than falllback to fixed IPs) or MAC blocking with the router which also would also prevent LAN control.

iointerrupt, 2022-10-02:
> Could setting up `cloudEnv` or another configuration parameter to a specific value prevent from attempt to query home? It would be more convenient that trying to block at the (local, e.g PiHole) DNS [...]

There is a parameter called `sDiscUrl` which I set to a local sinkhole server.  The default value is usually `https://disc-prod.iot.irobotapi.com/v1/robot/discover?robot_id={BLID}&country_code={COUNTRY_CODE}&sku={SKU_ID}`.  I usually set the parameter on my roombas (I have 3: 2 e5s and 1 i3) as follows: `{'topic':"wifictl", 'payload':'{ "state" : { "sdiscUrl" : "http://dev.null/roomba_discovery/" } }'}`.