koalazak/dorita980 #172: Enhancement: Roomba Now Supports At-Run-Time, By-Region Cleaning Options

shanemcw, 2023-02-01:
Data snippet below; should be possible to extend current cleaning API to support at-run-time cleaning options.

```
"lastCommand": {
  "command": "start",
  "initiator": "localApp",
  "ordered": 1,
  "pmap_id": "51xJPrplRCu53W5pqZHetg",
  "regions": [
   {
    "params": {
     "carpetBoost": false,
     "gentleMode": 0,
     "noAutoPasses": true,
     "twoPass": true,
     "vacHigh": true
    },
    "region_id": "0",
    "type": "zid"
   },
   {
    "params": {
     "carpetBoost": false,
     "gentleMode": 0,
     "noAutoPasses": true,
     "twoPass": true,
     "vacHigh": true
    },
    "region_id": "3",
    "type": "zid"
   },
   {
    "params": {
     "carpetBoost": false,
     "gentleMode": 0,
     "noAutoPasses": true,
     "twoPass": true,
     "vacHigh": true
    },
    "region_id": "1",
    "type": "zid"
   }
  ],
  "time": 1675171473,
  "user_pmapv_id": "230127T195526"
 }
```

shanemcw, 2023-02-04:
I plan to investigate. If I get it working, and working stably, I'll Pull Request for a Beta.

pix3d4, 2024-09-02:
Any news from this enhancement?
It would be very nice if this enhancement were implemented.
I have a question: 
How did you get the parameters that is possible to send in addition to the regions? i.e: twoPass, vacHigh and so on.
I have a roomba j7 combo and I can't figure out how to launch a vacuum or vaccum and mop.
Thanks