koalazak/dorita980 #174: Example getting "TypeError: dorita980.Local is not a constructor"

s28400, 2023-04-08:
Hey all, I am trying to use this tool but I am getting blocked with the following error after following the [Quick Start](https://github.com/koalazak/dorita980#quick-start-via-local-request-on-your-lan)

I receive the error:  "TypeError: dorita980.Local is not a constructor" on the var myRobotViaLocal = new dorita980.Local() line.

OS: Ubuntu 22.04
Node version: v12.22.9
NPM version: 8.5.1

Can anyone point me in the right direction with this? Thanks!

rahmanonik18, 2023-06-12:
Have you found the solution? I am stuck with this too, please let me know

bloomkd46, 2023-07-03:
Don't use `new`, just use `var myRobotViaLocal = dorita980.Local()`

koalazak, 2023-07-07:
WTF?
There is nothing wrong with `var myRobotViaLocal = new dorita980.Local()` in plain javascript node&12
Local() is a function

closed 2023-07-07 by koalazak