koalazak/dorita980 #175: npm audit -> Please update dependency

Mannshoch, 2023-05-08:
On my last update of node red I did an `npm audit` and received follow error message:
```
glob-parent  <5.1.2
Severity: high
glob-parent before 5.1.2 vulnerable to Regular Expression Denial of Service in enclosure regex - https://github.com/advisories/GHSA-ww39-953v-wcq6
No fix available
node_modules/glob-parent
  glob-stream  5.3.0 - 6.1.0
  Depends on vulnerable versions of glob-parent
  node_modules/glob-stream
    help-me  1.0.0 - 1.1.0
    Depends on vulnerable versions of glob-stream
    node_modules/help-me
      mqtt  1.14.1 - 4.2.6
      Depends on vulnerable versions of help-me
      node_modules/mqtt
        dorita980  >=1.0.1
        Depends on vulnerable versions of mqtt
        Depends on vulnerable versions of request
        Depends on vulnerable versions of request-promise
        node_modules/dorita980
          node-red-contrib-roomba980-fw2  *
          Depends on vulnerable versions of dorita980
          node_modules/node-red-contrib-roomba980-fw2

```

Could you please update your add-on - if possible? 

I use your tool by using https://github.com/yuchangyuan/node-red-contrib-roomba980-fw2.
See also [Issue 8](https://github.com/yuchangyuan/node-red-contrib-roomba980-fw2/issues/8)

Mannshoch, 2023-07-07:
Ping @koalazak

koalazak, 2023-07-07:
hi @Mannshoch 
Most of them are due to devDependencies (mocha, eslint, etc), so they do not affect you as a user. Only two are due to real dependencies and they do not seem relevant enough to spend time upgrading and resolving breaking changes. There's nothing to worry about. Having said that, they won't be resolved, but PRs are welcome. 

Thanks for reporting.

MaartenProvo, 2023-11-26:
:( My roomba stopped working also.

MaartenProvo, 2023-12-28:
What if we were to setup some for of crowd funding or financial contributions? Is there anyway we can pay someone to have a look and fix it?

closed 2024-03-11 by koalazak