koalazak/dorita980 #178: Issue retrieving password Cloud or local method

Bastien42, 2023-07-13:
i can't seem to retrieve the password of my roomba that wasn't the case somtime ago here is a log of the command run 

[image]

aelharda, 2023-07-14:
same error as the above

munkien, 2023-07-16:
Here too :(
iRobot I7 + clean base

[image]

bredowmax, 2023-07-17:
Same error message as @munkien reported

`Fatal error login into iRobot account. Please check your credentials or API Key.
An error occurred (AccessDeniedException) when calling the GetSecretValue operation: Access to KMS is not allowed`

jayvarner, 2023-07-17:
I also get the error described here. I was able to use this [Python Lib](https://github.com/NickWaterton/Roomba980-Python). There are instructions for [getting credentials from the cloud](https://github.com/NickWaterton/Roomba980-Python#getting-your-usernameblid-and-password-from-the-irobot-cloud). Apologies for suggesting a another project and not adding anything helpful to the issue here.

Bastien42, 2023-07-17:
> I also get the error described here. I was able to use this [Python Lib](https://github.com/NickWaterton/Roomba980-Python). There are instructions for [getting credentials from the [...]

yeah i tried this method too and this returns an error too

```
2023-07-17 17:09:42 INFO [Roomba.Password] Getting Roomba information from iRobot aws cloud...
Traceback (most recent call last):
  File "C:\Users\Bastien\Desktop\Roomba980-Python-master (1)\Roomba980-Python-master\roomba\password.py", line 294, in <module>
    main()
  File "C:\Users\Bastien\Desktop\Roomba980-Python-master (1)\Roomba980-Python-master\roomba\password.py", line 291, in main
    get_passwd.get_password()
  File "C:\Users\Bastien\Desktop\Roomba980-Python-master (1)\Roomba980-Python-master\roomba\password.py", line 117, in get_password
    from getcloudpassword import irobotAuth
  File "C:\Users\Bastien\Desktop\Roomba980-Python-master (1)\Roomba980-Python-master\roomba\getcloudpassword.py", line 20, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

FidgetyRat, 2023-07-17:
> I also get the error described here. I was able to use this [Python Lib](https://github.com/NickWaterton/Roomba980-Python). There are instructions for [getting credentials from the [...]

Don't apologize. Im sure most people here just want a solution and the python lib was that.

RJiraya, 2023-07-17:
Hello !

Did you retry today ? The problems seems to be fixed by iRobot team !

It's working on my side.

Bastien42, 2023-07-17:
well yes you're right whatever was the issue now it works with doitu980 too