koalazak/dorita980 #183: Install instructions unclear.

CrysKilljoy, 2023-11-20:


CrysKilljoy, 2023-11-20:
after trying around, you should not opne node.js directly but the command prompt. and you shall not use the "$" in the presented code so just paste "npm install dorita980 --save"

now i get 
`C:\Users\chrja>npm install dorita980 --save
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated request-promise@4.2.6: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

added 126 packages in 9s

4 packages are looking for funding
  run `npm fund` for details
npm notice
npm notice New minor version of npm available! 10.1.0 -> 10.2.4
npm notice Changelog: https://github.com/npm/cli/releases/tag/v10.2.4
npm notice Run npm install -g npm@10.2.4 to update!
npm notice`

koalazak, 2023-11-20:
hi @CrysKilljoy, this library is for node.js developers. For them the documentation is pretty clear.
regards,

closed 2023-11-20 by koalazak