koalazak/dorita980 #186: getRobotState last command shows regions as [Object] not full details

dmz006, 2024-03-09:
Running the following:
myRobotViaLocal.on('connect', init);

function init () {
  myRobotViaLocal.getRobotState(['batPct', 'bbchg3']).then((actualState) => {
    console.log(actualState);
    myRobotViaLocal.end();
  });
}

I get the full state details but lastCommand (when last command submitted cleaning request for all regions in the house, individually selected):
  lastCommand: {
    command: 'start',
    initiator: 'rmtApp',
    time: 1710026220,
    ordered: 1,
    pmap_id: 'y2PgGOh8REGTkpKjo2vdcg',
    regions: [
      [Object], [Object],
      [Object], [Object],
      [Object], [Object],
      [Object], [Object]
    ],
    user_pmapv_id: '230314T173508'
  },

This is on an s9+ - not sure easiest way to get firmware version info; but most command seem to run; and all other objects in stats expand out.

bloomkd46, 2024-03-10:
Try running 
```javascript
myRobotViaLocal.on('connect', init);

function init () {
myRobotViaLocal.getRobotState(['batPct', 'bbchg3']).then((actualState) => {
console.log(JSON.stringify(actualState, null, 2));
myRobotViaLocal.end();
});
}
```

you need to run `JSON.stringify()` to see the contents of the nested objects. (The extra `2` argument is to make the JSON easier to read)

dmz006, 2024-03-10:
That got it, thank you

closed 2024-03-10 by dmz006