koalazak/dorita980 #187: Issues installing

ScenicSystemsLLC, 2024-03-11:
When I try to install this in a docker image I am constantly getting errors and messages that functions have been deprecated I am unable to run the package.  Is there any advice or updates available.

koalazak, 2024-03-11:
Impossible to help without more information about the isse

closed 2024-03-11 by koalazak

ScenicSystemsLLC, 2024-03-11:
When I try to install using option 1 I get:

***@***.***:/home/bill# npm install -g dorita980
npm WARN deprecated ***@***.******@***.***>: this library is no longer supported
npm WARN deprecated ***@***.******@***.***>: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated ***@***.******@***.***>: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated ***@***.******@***.***>: request has been deprecated, see https://github.com/request/request/issues/3142

changed 125 packages in 5s

I cannot launch or run any docker commands.

I am running Ubuntu desktop 23.10
Installed the latest Dockor, NPM, and Nodejs

Bill Welch
Senior Systems Engineer
Scenic Systems
803-741-4436
***@***.******@***.***>

***@***.***[Graphical user interface, application  Description automatically generated][A picture containing icon  Description automatically generated][A yellow sign with black text  Description automatically generated with low confidence]
***@***.***

koalazak, 2024-03-11:
there is no issue with that installation. all warnings.