koalazak/dorita980 #188: Out of date dependencies

clintongormley, 2024-03-11:
Hiya

I've just tried following the instructions to clone the git repo and to install using `npm`. I can't proceed because of some dependencies which cannot be fixed:

```
npm install
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated request-promise@4.2.6: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated circular-json@0.3.3: CircularJSON is in maintenance only, flatted is its successor.
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated json3@3.3.2: Please use the native JSON object instead of JSON 3
npm WARN deprecated istanbul@0.4.5: This module is no longer maintained, try this instead:
npm WARN deprecated   npm i nyc
npm WARN deprecated Visit https://istanbul.js.org/integrations for other alternatives.

added 368 packages, and audited 369 packages in 22s

18 packages are looking for funding
  run `npm fund` for details

21 vulnerabilities (11 moderate, 7 high, 3 critical)

To address issues that do not require attention, run:
  npm audit fix

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
clintons-mbp:dorita980 clinton$ npm audit fix --force
npm WARN using --force Recommended protections disabled.
npm WARN audit Updating request-promise to 0.0.1, which is a SemVer major change.
npm WARN audit Updating eslint to 8.57.0, which is a SemVer major change.
npm WARN audit Updating mocha to 10.3.0, which is a SemVer major change.
npm WARN audit Updating request-promise to 0.0.1, which is a SemVer major change.
npm WARN audit Updating semistandard to 17.0.0, which is a SemVer major change.
npm WARN audit Updating mqtt to 5.3.6, which is a SemVer major change.
npm WARN deprecated request-promise@0.0.1: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142

added 226 packages, removed 166 packages, changed 67 packages, and audited 429 packages in 24s

123 packages are looking for funding
  run `npm fund` for details

# npm audit report

request  *
Severity: moderate
Server-Side Request Forgery in Request - https://github.com/advisories/GHSA-p8p7-x288-28g6
Depends on vulnerable versions of tough-cookie
No fix available
node_modules/request

tough-cookie  <4.1.3
Severity: moderate
tough-cookie Prototype Pollution vulnerability - https://github.com/advisories/GHSA-72xf-g2v4-qvf3
No fix available
node_modules/tough-cookie

2 moderate severity vulnerabilities

Some issues need review, and may require choosing
a different dependency.
clintons-mbp:dorita980 clinton$ npm audit fix --force
npm WARN using --force Recommended protections disabled.
npm WARN audit No fix available for request@*

added 1 package, and audited 430 packages in 1s

123 packages are looking for funding
  run `npm fund` for details

```

koalazak, 2024-03-11:
why you can't procedure? I only see warnings there. no errors.

closed 2024-03-11 by koalazak

clintongormley, 2024-03-11:
I don't know `npm` well enough so maybe I'm missing something obvious, but the package doesn't install. It just tells me that there is a problem without a fix:

```
npm install --force
npm WARN using --force Recommended protections disabled.

up to date, audited 430 packages in 1s

123 packages are looking for funding
  run `npm fund` for details

2 moderate severity vulnerabilities

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
```

If it is just a warning, how should I proceed? Using `--force` doesn't  seem to do it

koalazak, 2024-03-11:
I see there that the installation was success

clintongormley, 2024-03-11:
Ah! My apologies. My assumptions were incorrect

thanks for the help