koalazak/dorita980 #189: Can't connect to get blid.

h00pak, 2024-03-14:
Using get-roomy-password <ip address> gives result of unsafe legacy negotiation disabled and trying to get-roomy-password-cloud <login> <password> is now saying that the API Key is required.

Any way around these? How does one get an API Key?

kkaminski7, 2024-03-24:
I am experiencing exactly the same the same issue.

h00pak, 2024-03-24:
For me, this ended up being a problem with two special characters in a row at the end of my password. I was using the Nick Waterton solution and finally changed the password on a guess. Immediately returned all the data.

kkaminski7, 2024-03-24:
Thank You h00pak for update on Your situation. It seems like we had the same issue, I have changed password to more human readable form and it works for me now aswell.

closed 2024-03-24 by koalazak