koalazak/dorita980 #193: get-roomba-password-cloud gives authentication errors

Lindahl, 2024-10-12:
I'm trying to use get-roomba-password-cloud, but I'm getting authentication errors. I'm using the same password and username I use to log into the app on a phone, or on the iRobot website. Did they migrate off of Gigya, or do I need a different Gigya API key than the default one?

$ get-roomba-password-cloud <username> <password>
Authentication error. Please check your credentials.
{
  callId: 'fb8396a9bafb4fdeb7809a02bc4935fb',
  errorCode: 403042,
  errorDetails: 'invalid loginID or password',
  errorMessage: 'Invalid LoginID',
  apiVersion: 2,
  statusCode: 403,
  statusReason: 'Forbidden',
  time: '2024-10-12T13:33:12.152Z'
}

sbogomolov, 2024-11-30:
I'm having the same issue. The command I run:
```
docker run --rm -it node sh -c "npm install -g dorita980 && get-roomba-password-cloud 'user@domain.tld' 'password'"
```
result:
```
Authentication error. Please check your credentials.
{
  callId: '4024656685a547879f41e8c7dae39c97',
  errorCode: 403042,
  errorDetails: 'invalid loginID or password',
  errorMessage: 'Invalid LoginID',
  apiVersion: 2,
  statusCode: 403,
  statusReason: 'Forbidden',
  time: '2024-11-30T19:24:03.553Z'
}
```

The same email and password work fine to login via mobile app and on https://www.irobot.com/.

sbogomolov, 2024-12-02:
`password.py` from this repository worked for me: https://github.com/NickWaterton/Roomba980-Python.

scottbrady, 2025-02-18:
I had this same error and fixed it by changing my iRobot password to NOT contain special characters. I think this utility's argument parsing gets confused by any special characters in the password (like dashes etc).

Steps:

1) Change your iRobot password to something without special characters. For example, just use letters and numbers
2) Run `get-roomba-password-cloud`
3) Change your iRobot password back to something stronger

Hope that helps.

Delfairen, 2025-04-26:
> I had this same error and fixed it by changing my iRobot password to NOT contain special characters. I think this utility's argument parsing gets confused by any special characters in the password [...]

This resolved the issue for me although it is worth noting that the app and website both require a special character so just need to use a special character that Linux does not use eg ! and &

marksverdhei, 2025-05-21:
> I had this same error and fixed it by changing my iRobot password to NOT contain special characters. I think this utility's argument parsing gets confused by any special characters in the password [...]

Thank you! Yeah, it would be nice if the documentation mentioned this as a heads up. It's easy to forget accounting for special characters in a bash terminal when pasting in some password.