koalazak/dorita980 #198: Adding the Irobot J5 model

Tazmanian79, 2024-12-02:
I have a Irobot J5 which I want to add to my home assistant.
I used to have an older model which was working fine in HA but the robot broke.
So I replaced it with the J5 model.

But I’m not able to add him. The integration says it cannot retrieve the password.

I also ran following command:

docker run -it node sh -c "npm install -g dorita980 && get-roomba-password <robotIP>"
And I get this info but it does not get the password. What is wrong?

Robot Data:
{
ver: ‘4’,
hostname: ‘iRobot-85CCAC5D51D54B8D8D5803B03BB1FAA8’,
robotname: ‘Dusty’,
robotid: ‘xxx’,
ip: ‘xxx’,
mac: ‘xxx’,
sw: ‘xxx’,
sku: ‘j517840’,
nc: 0,
proto: ‘mqtt’,
cap: {
binFullDetect: 2,
addOnHw: 3,
oMode: 10,
odoa: 4,
dockComm: 1,
maps: 3,
pmaps: 9,
mc: 2,
tLine: 2,
area: 1,
eco: 1,
multiPass: 2,
edge: 0,
pp: 0,
team: 1,
pose: 2,
lang: 2,
hm: 0,
‘5ghz’: 1,
prov: 3,
sched: 2,
svcConf: 1,
ota: 2,
log: 2,
langOta: 0,
ns: 1,
expectingUserConf: 2
},
cloudConnState: 14,
blid: ‘xxx’
}
Error getting password. Follow the instructions and try again.

Tazmanian79, 2025-01-06:
Anyone?

Tichael, 2025-03-16:
I was also having issue with my Roomba J5 when running the command on Docker on Windows. Installing Node.js LTS locally (at the time of writing this, it's v22.14.0) and doing:
```
$ npm install -g dorita980
$ get-roomba-password-cloud "<iRobot Username>" "<iRobot Password>"
```

Worked perfectly fine. I also had to put double quotes around my username and password, especially since my password uses special character that a shell missunderstand.

closed 2025-11-05 by Tazmanian79